qmk/qmk_firmware #11755: [Bug] PrimeKB/Meridian LED indicator lights won't turn off.

newtonapple, 2021-02-01:
[Meridian](https://github.com/qmk/qmk_firmware/tree/master/keyboards/primekb/meridian)'s LED indicator lights won't turn off.  Further more, it seems like `rgblight_setrgb_at` can no longer control the LEDs, which was previously working in the [initial commit](https://github.com/qmk/qmk_firmware/commit/f079dd90ed88fd29b7e5bbdd3496742f3afeff4e).

## Describe the Bug

It seems like `rgblight_setrgb_at` no longer works properly for the Meridian firmwares (both VIA & default) on master.  I'm not sure what could have caused it or if other STM32 related PCBs have the same issue.  After git-bitsecting, I have narrowed it down to [this commit](https://github.com/qmk/qmk_firmware/commit/c66df1664497546f32662409778731143e45a552) to be the source of the problem.

## System Information

 - Keyboard:  
   - Revision (if applicable):  [PrimeKB/Meridian](https://github.com/qmk/qmk_firmware/tree/master/keyboards/primekb/meridian) (revision master)
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: `avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0`
 - arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.11.56
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x ] Karabiner
   - [ ] Other:

## Additional Context

- `git-bitsect` points to this commit to be the source of the problem: https://github.com/qmk/qmk_firmware/commit/c66df1664497546f32662409778731143e45a552.

fauxpark, 2021-02-01:
Can't see anything in the commit list for the last breaking changes PR (https://github.com/qmk/qmk_firmware/pull/11053/commits) that stands out to me. There doesn't seem to have been much activity near the RGBLight stuff this round. I'm thinking it may be something to do with the ChibiOS config files. @tzarc?

alvinsee, 2021-02-01:
The offending PR seems to be [this change](https://github.com/qmk/qmk_firmware/pull/10214). 
It is not apparent to me that `platforms/chibios/GENERIC_STM32_F072XB/board/board.c` is equivalent to `lib/chibios/os/hal/boards/ST_NUCLEARO64_F072RB`. 

Steps to revert to intended functionality:
```
git checkout f079dd9 -- platforms/
make primekb/meridian:via:flash
```

fauxpark, 2021-02-01:
They don't seem to be equivalent. `board.c` appears to be functionally the same, accounting for whitespace/formatting.

`board.h` however has significant differences. In particular the `STM32_HSECLK` defaults to `8000000U`, but was `0U` in the deleted board.h. I'm not sure if the SPI peripheral uses this.

alvinsee, 2021-02-01:
I tried grepping for `STM32_HSECLK` but failed to find any usages. I've also tried adding the value `0U` to `config.h`, but that seems moot.

sigprof, 2021-10-03:
Looking at the message dates here and in #11812, I see that the attempt to fix this board by switching to the bitbang driver was apparently made before PR #13022 was merged, which should have brought the code mostly to the same state as it was before those ChibiOS changes (the problem was that SPI on STM32F072 requires assigning SCK to a physical pin to have a functional SPI receiver, and the SPI driver relies on the receiver part for transfer completion; however, the board defaults for F072 were changed to no longer assign SPI2_SCK to a physical pin by default).

But the Meridian code has one more potential problem that might show up when using the WS2812 SPI driver — it issues several `rgblight_setrgb_at()` calls without any delay between them. The WS2812 SPI driver may not handle such calls properly with the default options. However, you may try adding the following code to `config.h`:
```c
#define WS2812_SPI_USE_CIRCULAR_BUFFER
```
This makes the WS2812 SPI driver behave similar to the PWM driver — it will repeat the bitstream continuously, and any `rgblight_setrgb_at()` calls will just change the bitstream data in the RAM buffer without even touching the SPI controller.

Alternatively, you can try another option:
```c
#define WS2812_SPI_SYNC
```
This option makes the WS2812 SPI driver wait until the transfer is completed before returning (similar to the bitbang driver behavior, except interrupts are not disabled during all that time), therefore the error condition (starting a new SPI transfer before the previous one has completed) is avoided. However, this mode stops the normal keyboard processing while the bitstream is sent (which may be really noticeable with larger chains), and if the SPI completion is not detected properly (e.g., because the SCK output is not mapped to a physical pin), the driver will just hang indefinitely, so this mode should probably be used only for testing.

@holtenc @Woovie Could you test these options on the real hardware instead of changes suggested in #11812 or #14678?

Woovie, 2021-10-03:
I appreciate you digging into it @sigprof, We will see if works.