qmk/qmk_firmware #11763: Avoid Tap Dance delay if only single tap and long press actions set

andreyvit, 2021-02-01:
If only single tap and long press actions are set for a Tap Dance key, we can execute its action as soon as it is released without waiting for the tapping term.

## Feature Request Type

- [x] Alteration (enhancement/optimization) of existing feature(s)

## Description

Currently, if I set e.g. C to tap dance with single press giving C and long press giving Command-C, the keyboard delays processing the press even after I release the key. This is unnecessary and significantly worsens the user experience, to the point of being unusable (especially for keys where immediate feedback is important, like Backspace).

When a tap dance key is released, I'd suggest executing the action immediately if all the programmed taps have already occurred.

andreyvit, 2021-03-01:
Update: this has been fixed some time since opening the issue; now works as expected.

fandy, 2021-08-14:
This still seems to be an issue on the Moonlander keyboard. Tapdancing results in the tap stroke registering on keyup, instead of keydown, which gives the perception of a delay.

dirk-bester, 2023-01-03:
3> This still seems to be an issue on the Moonlander keyboard. Tapdancing results in the tap stroke registering on keyup,

> This still seems to be an issue on the Moonlander keyboard. Tapdancing results in the tap stroke registering on keyup, instead of keydown, which gives the perception of a delay.

What you describe is wrong and has nothing to do with the fixed bug. You cannot immediately do the tap on key down, it can only ever be on key up. How will the hold happen if you already did the tap?

drashna, 2023-02-06:
> Update: this has been fixed some time since opening the issue; now works as expected.

What was the fix here?

andreyvit, 2023-02-07:
@drashna The fix was executing the single tap action immediately on key up. Don't know the particular commit.

The original issue has been fixed and has not come back; please open a separate ticket if you have a different issue.

closed 2023-02-07 by andreyvit

saladpanda, 2023-02-08:
@drashna 

I just tried compiling a keymap generated by Oryx against upstream qmk_firmware (it needed a little rewriting, but not too much) and this issue hasn't been fixed in QMK.
The reason is probably, that this is not a bug in QMK in the first place. It's rather a flaw in the keymap.c which is generated by Oryx.

What Oryx generates for a tap-dance-key is a line like
```
[DANCE_1] = ACTION_TAP_DANCE_FN_ADVANCED(on_dance_1, dance_1_finished, dance_1_reset),
```
with the following function definitions:
```c
void on_dance_1(qk_tap_dance_state_t *state, void *user_data) {
    if(state->count == 3) {
        tap_code16(KC_2);
        tap_code16(KC_2);
        tap_code16(KC_2);
    }
    if(state->count > 3) {
        tap_code16(KC_2);
    }
}

void dance_1_finished(qk_tap_dance_state_t *state, void *user_data) {
    dance_state[1].step = dance_step(state);
    switch (dance_state[1].step) {
        case SINGLE_TAP: register_code16(KC_2); break;
        case SINGLE_HOLD: register_code16(KC_F2); break;
        case DOUBLE_TAP: register_code16(KC_2); register_code16(KC_2); break;
        case DOUBLE_SINGLE_TAP: tap_code16(KC_2); register_code16(KC_2);
    }
}

void dance_1_reset(qk_tap_dance_state_t *state, void *user_data) {
    wait_ms(10);
    switch (dance_state[1].step) {
        case SINGLE_TAP: unregister_code16(KC_2); break;
        case SINGLE_HOLD: unregister_code16(KC_F2); break;
        case DOUBLE_TAP: unregister_code16(KC_2); break;
        case DOUBLE_SINGLE_TAP: unregister_code16(KC_2); break;
    }
    dance_state[1].step = 0;
}
```

The "correct" way, which would solve the flaw discussed in this issue, is actually given as an example in the QMK Documentation:
https://docs.qmk.fm/#/feature_tap_dance?id=example-3

To cite the code from that example:
```c
typedef struct {
    uint16_t tap;
    uint16_t hold;
    uint16_t held;
} tap_dance_tap_hold_t;

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    qk_tap_dance_action_t *action;

    switch (keycode) {
        case TD(CT_CLN):  // list all tap dance keycodes with tap-hold configurations
            action = &tap_dance_actions[TD_INDEX(keycode)];
            if (!record->event.pressed && action->state.count && !action->state.finished) {
                tap_dance_tap_hold_t *tap_hold = (tap_dance_tap_hold_t *)action->user_data;
                tap_code16(tap_hold->tap);
            }
    }
    return true;
}

void tap_dance_tap_hold_finished(qk_tap_dance_state_t *state, void *user_data) {
    tap_dance_tap_hold_t *tap_hold = (tap_dance_tap_hold_t *)user_data;

    if (state->pressed) {
        if (state->count == 1
#ifndef PERMISSIVE_HOLD
            && !state->interrupted
#endif
        ) {
            register_code16(tap_hold->hold);
            tap_hold->held = tap_hold->hold;
        } else {
            register_code16(tap_hold->tap);
            tap_hold->held = tap_hold->tap;
        }
    }
}

void tap_dance_tap_hold_reset(qk_tap_dance_state_t *state, void *user_data) {
    tap_dance_tap_hold_t *tap_hold = (tap_dance_tap_hold_t *)user_data;

    if (tap_hold->held) {
        unregister_code16(tap_hold->held);
        tap_hold->held = 0;
    }
}

#define ACTION_TAP_DANCE_TAP_HOLD(tap, hold) \
    { .fn = {NULL, tap_dance_tap_hold_finished, tap_dance_tap_hold_reset}, .user_data = (void *)&((tap_dance_tap_hold_t){tap, hold, 0}), }

qk_tap_dance_action_t tap_dance_actions[] = {
    [CT_CLN] = ACTION_TAP_DANCE_TAP_HOLD(KC_COLN, KC_SCLN),
};
```

I implemented this in my keymap and it works exactly as expected. The tap-keycode is sent as soon as the key is released and the hold-keycode triggers after the key is held for the TAPPING-TERM. **Problem solved**.

If you (or someone else at ZSA) could adjust Oryx to generate code accordingly, that would be awesome.

KaelWD, 2023-12-19:
If anyone else finds this, turning off "Prefer tap action in Tap-Hold tap dances" fixed it for me. The description is incorrect, it actually sends the tap action immediately when rolled over even with the setting off.

whaaaley, 2023-12-31:
@KaelWD what other settings do you have enabled/disabled in tapping and dual functions settings? i think there must be a combination of settings i'm missing.

my setup is, i'm using a dual function key that is "held" and "tap and held" so there's no need for any delay here for me.

KaelWD, 2023-12-31:
I have "Prefer tap action in Tap-Hold tap dances" off and "Tapping force hold" on

> i'm using a dual function key that is "held" and "tap and held"

Where is the delay then? The problem I was having was a delayed tap action on keys that also have hold.