qmk/qmk_firmware #11794: [Bug] brew qmk/qmk/qmk does not build on M1 Mac

jknotzke, 2021-02-04:
Using brew 2.7.7 calling "brew install qmk/qmk/qmk ends in an error:

brew install qmk/qmk/qmk
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
==> Updated Casks
Updated 5 casks.

==> Installing qmk from qmk/qmk
avr-gcc@8: The x86_64 architecture is required for this software.
Error: An unsatisfied requirement failed this build.

## Describe the Bug

The brew cask appears to be broke when on an M1 (ARM based) Mac using MacOS 11.2

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-02-04:
Issue is with avr-gcc, not qmk.
https://github.com/osx-cross/homebrew-avr

closed 2021-02-04 by jknotzke

dgdosen, 2021-03-06:
Before I recompile - Does anyone know if that means that you could compile firmware to run on either "arm64" OR 'x86_64"?

fauxpark, 2021-03-06:
The firmware does not run on either - arm64 and x86_64 are the architectures of the host system, ie. the compiler itself. The target architecture will be either `avr` or one of several subsets of ARM designed specifically for microcontrollers. QMK cannot run on the host, as it is fairly tied to some aspects of microcontrollers that a full x86_64 system, for example, would not have. However, there is a vague plan for a "libqmk" which would abstract some of the core of QMK out into something that could potentially run on the host: https://github.com/qmk/libqmk

mmshivesh, 2021-12-07:
You can build avr-gcc from source as they discuss in this issue https://github.com/osx-cross/homebrew-avr/issues/237