qmk/qmk_firmware #11818: [Bug] avr-ld libctf.so.0 not found

leshow, 2021-02-07:
## Describe the Bug

I have been using QMK for years, recently I did a fresh checkout and was unable to build any firmware, it exits with this error:

```
~/dev/ergo/qmk master
❯ make crkbd:default              
QMK Firmware 0.11.62
Making crkbd/rev1/legacy with keymap default

avr-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/crkbd_rev1_legacy_default.elf                                                       [ERRORS]
 | 
 | /usr/bin/avr-ld: error while loading shared libraries: libctf.so.0: cannot open shared object file: No such file or directory
 | collect2: error: ld returned 127 exit status
 | 
make[1]: *** [tmk_core/rules.mk:320: .build/crkbd_rev1_legacy_default.elf] Error 1
Make finished with errors
make: *** [Makefile:523: crkbd:default] Error 1
```

## System Information

 - Keyboard:  
crkbd
 - Operating system:        
Linux leshowbox 5.10.13-arch1-2 #1 SMP PREEMPT Sat, 06 Feb 2021 11:07:29 +0000 x86_64 GNU/Linux
 - AVR GCC version: 
avr-gcc (GCC) 8.3.0
 - ARM GCC version: 
arm-none-eabi-gcc (Arch Repository) 10.2.0
 - QMK Firmware version:
0.11.62

Googling has not returned any good results. I assume this broke from an update to some arch dependency as even my old QMK checkout at version 0.7.53 exits with the same error

leshow, 2021-02-07:
I see there is this issue in the arch bug tracker, so that's probably it https://bugs.archlinux.org/task/69567

closed 2021-02-07 by leshow

agarciamontoro, 2021-02-10:
I experienced the same issue, and fixed it downgrading the binutils package doing the following:

```
asp checkout binutils
cd binutils
git checkout 3647ee5
cd trunk
makepkg -sci
```

I succesfully flashed my keyboard after this.

prototype00, 2021-02-12:
I clean install Arch linux, so there is not old pkg now. When is this fix it ?

ivanjermakov, 2021-02-22:
In addition to @agarciamontoro answer, you may need to specify `--skippgpcheck` for the `makepkg` command, since some signatures are not verifying.
```
makepkg -sci --skippgpcheck
```

leshow, 2021-02-22:
I just used [downgrade](https://aur.archlinux.org/packages/downgrade/) specify any version at or below `2.35-2` and you should be alright

willpnw, 2021-02-25:
I followed the recommendation of @leshow It was very quick, very easy, and worked immediately

rktjmp, 2021-02-27:
NB: You can use docker to build, avoiding any potential hassles with pacman updates re-breaking things or your distro getting out of sync.

`Dockerfile.build_local`:
```
FROM qmkfm/base_container AS build-stage

VOLUME /qmk_firmware
WORKDIR /qmk_firmware
COPY . .

RUN make <manu>/<keeb>/<reev>:<you>

FROM scratch AS export-stage
COPY --from=build-stage /qmk_firmware/.build/*.hex /
```

Then run

`DOCKER_BUILDKIT=1 docker build -o hex -f Dockerfile.build_local` and your firmware will be in `<qmk_root>/hex`.

Probably be a bit slower to build since it has to copy a bunch of files into the container. You could get around this by setting up a bind mount and running the container instead of building it, but that's more than a one liner.

pcjentsch, 2021-02-28:
The workaround in this thread was working fine until today, where I get the error
```
 | /usr/bin/avr-ld: /usr/lib/libctf.so.0: version `LIBCTF_1.1' not found (required by /usr/bin/avr-ld)
 | collect2: error: ld returned 1 exit status
 ```
anyone know how to address this? I figure the package `avr-ld` is now too new to be able to use the older libctf.

pcjentsch, 2021-02-28:
just downgrading `avr-binutils` to 2.35 with pacman worked fine, should've tried that first!

sigprof, 2021-03-02:
Looks like the real bug is in the `avr-binutils` Arch package (even though the `binutils` package no longer provides `libctf.so.0`, the `avr-binutils` package is built with some options that make all executables depend on that library): https://bugs.archlinux.org/task/69861

Downgrading just the `binutils` package to 2.35 helped when the `avr-binutils` package was also 2.35, but the latest upgrade to `avr-binutils` introduced a dependency on a newer `libctf.so.0` version from 2.36, which does not exist in any Arch package, therefore both `binutils` and `avr-binutils` need to be downgraded now, until that bug is finally fixed.

rstacruz, 2021-03-07:
Hint for Arch Linux users who may have landed here looking for a solution:

```sh
# You may have some of the old versions still in your Pacman cache
ls /var/cache/pacman/pkg/avr-binutils-2.35.1-1-x86_64.pkg.tar.zst /var/cache/pacman/pkg/binutils-2.35.1-1-x86_64.pkg.tar.zst

# Install them if they are
sudo pacman -U /var/cache/pacman/pkg/avr-binutils-2.35.1-1-x86_64.pkg.tar.zst /var/cache/pacman/pkg/binutils-2.35.1-1-x86_64.pkg.tar.zst
```

Otherwise, there's https://wiki.archlinux.org/index.php/Downgrading_packages for other methods of obtaining those versions.

rktjmp, 2021-03-07:
AFAIK the latest package has fixed the issue, mine built without issue a few hours ago.

areThereAnyUserNamesLeft, 2021-03-20:
Yep - as @rktjmp commented ⬆️ -this was resolved with a recent update for me too