qmk/qmk_firmware #11868: GamaKay MK61 support

skibbipl, 2021-02-12:
## Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Hi, I recently bought GamaKay MK61 keyboard. I really like it so far, but the lack of powerful firmware is hurting me :(
Yesterday I've opened it and from what I can see it runs on Weltrend WT32L064 ARM chip with two AW9523B LED controllers. I can see this weltrend chip is pretty well documented, so perhaps it would be possible to port QMK to it? I'm not that smart to understand the docs, but perhaps some experts can comment on this?
http://www.weltrend.com/en-global/product/detail/67/118/524
http://www.weltrend.com.tw/upload/website/product/WT32L064-032_DataSheet_EN_V04(20200908).pdf
http://www.weltrend.com/upload/media/Doc/Support/WT32L064_sample_2020820.7z
https://www.awinic.com/en/index/pageview/catid/19/id/15.html

marcovirtual, 2021-04-14:
Really would like to see the Gamakay MK61 being ported to QMK also.

SeerLite, 2021-07-23:
I extracted the files from Gamakay's software updater. It's an Electron app so I just had to look inside `app.asar`. I found the .hex file for the MK61 in a folder named "MK61L" and also one for another keyboard, in a folder named "MK68M" (not sure what model that one is though, I guess T68?). I'm not sure if I'm allowed to post them here or if they're useful.

I also looked inside the actual JS of the Electron app. I have a rough idea of what the updater does to write the hex data to the keyboard. But, again, I'm not sure how useful that is so I haven't gotten that deep into it.
What I know is that at one point a file with .node extension is imported and that acts as the interface for the JS and the hardware (HID device?). `file` tells me it's a Windows DLL, so I guess that checks out?
Fortunately, all the logic behind writing the data seems to be in the JS, which is also as deep as I can get for now.
There's also some JS arrays labeled "matrix" with hexadecimal codes and others with the symbols (like "esc", "tab" and such) which I guess could come really useful? or maybe not at all?

Or maybe the actual important stuff is all inside the .hex files? I know very little about low-level stuff like this so I don't even know where to start to read them. I also have no idea how to interpret the datasheets posted above, and much less how to actually write a port using them. Now, if any expert is willing to give me some starting points and hints to what I should look for, maybe I can give it a try (or do the easy parts of the process), assuming it's not too confusing and I don't give up immediately lol. I'd love to learn more about this, it's just that I don't know where to start.

Otherwise I'd still be very grateful if an expert took over and got this working. Just ask about anything you need to know about the files and I'll do my best to give the necessary information.

The updater can be downloaded from the Banggood listing for the MK61 under Specification and then Precise details, but I can provide with all the files if needed.

kb-elmo, 2021-07-28:
To port QMK to this keyboard support for this chip would first have to be implemented on upstream ChibiOS since that is what QMK uses to run on ARM based chips.  
For that someone would have to volunteer as a maintainer for that ChibiOS port.

Directly porting QMK to a new ARM chip is not possible and as long as there is no support for it on ChibiOS this board won't get QMK support.

> I also looked inside the actual JS of the Electron app. I have a rough idea of what the updater does to write the hex data to the keyboard. But, again, I'm not sure how useful that is so I haven't gotten that deep into it.

Third party configurator apps are not of any interest for QMK as it is highly unlikely that any kind of support for those apps is going to be implemented into QMK.
For QMK other configurator programs (VIA and Vial) already exist that use their own protocols.

closed 2022-01-09 by stale[bot]