qmk/qmk_firmware #11925: [Bug] qmk setup throws error asks for MSYS2 MinGW 64-bit terminal even when it's already running there.

henhen724, 2021-02-15:
I am trying to install QMK on windows 10 using an MSYS2 terminal.  I have installed mingw-w64-x86_64-toolchain and mingw-w64-x86_64-python3-pip.  I am now trying to run `qmk setup` in the  #MSYS2 MinGW 64-bit terminal.  I have also set the SHELL environment variable to "/usr/bin/bash" using my .bashrc following the instructions from issue #9950.

## Describe the Bug
I am running `qmk setup` in a MSYS2 MinGW 64-bit terminal and getting the following error:
```
henhe@Henrys-NYC-Desktop MINGW64 ~
$ qmk setup
ERROR: It seems you are not using the MINGW64 terminal.
Please close this terminal and open a new MSYS2 MinGW 64-bit terminal.
Python: /usr/bin/python3.exe, MSYSTEM: MINGW64
```

## System Information
 - Operating system: Windows 10 NT
 - Any keyboard related software installed?  No

fauxpark, 2021-02-15:
See my last comment on that issue.

henhen724, 2021-02-16:
Solution: Make sure C:/msys64/mingw64/bin is included in your PATH and is included before C:/msys64/usr/bin. 

In addition, your terminal will not recognize the command qmk as mingw64/bin/qmk.exe.  It will only recognize qmk.exe.  So write:
`
qmk.exe setup
`
and not
`
qmk setup
`

fauxpark, 2021-02-16:
Nope, that's not how it should be. Nuke your MSYS install (including the folder) and start again, this time make sure you install the MinGW64 Python before installing the QMK CLI.

mianqi2016, 2021-11-08:
Found the reason: "pythons-pip" is not "mingw-w64-x86_64-python3-pip".

Solution:

1. After installed MSYS2, don't run it.
2. Open MSYS2 MinGW 64-bit(or 32-bit).
3. Install mingw-w64-x86_64-python3-pip, git, mingw-w64-x86_64-toolchain there.
4. Run "python3 -m pip install qmk" inside MSYS2 MinGW 64-bit(or 32-bit) also.

koron, 2023-09-16:
Please try with two environment variables with values:

```
export SHELL= MSYSTEM=MINGW64
```

fauxpark, 2023-09-17:
That is not the correct solution. Please use QMK MSYS instead of vanilla MSYS2: https://msys.qmk.fm.

closed 2023-09-17 by fauxpark