qmk/qmk_firmware #12071: Hi, do you know why the layout in the QMK Configurator does not match that in the keymap.c file?

YangPiCui, 2021-03-01:
Hi, do you know why the layout in the QMK Configurator does not match that in the keymap.c file?
https://config.qmk.fm/#/handwired/evk/v1_3/LAYOUT
https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/evk/v1_3/keymaps/default/keymap.c

[image]
[image]

_Originally posted by @YangPiCui in https://github.com/qmk/qmk_firmware/issues/11880#issuecomment-787069007_

YangPiCui, 2021-03-01:
v1_3.h looks fine, too:
[image]
https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/evk/v1_3/v1_3.h

fauxpark, 2021-03-01:
Default keymaps for the Configurator are added here: https://github.com/qmk/qmk_configurator/tree/master/public/keymaps

It does not use the `default` keymap.c from the qmk_firmware repo.

YangPiCui, 2021-03-03:
Hi, 
My changes to the .json file has been commited.
https://github.com/qmk/qmk_configurator/pull/891

Do you know why the keymap still hasn't been updated on the QMK Configurator?
https://config.qmk.fm/#/handwired/evk/v1_3/LAYOUT

noroadsleft, 2021-03-03:
The deployment task had a failed test unrelated to your changes.

I've restarted the deployment task.

noroadsleft, 2021-03-03:
@YangPiCui,

Okay, deployment succeeded, but the keymaps still isn't rendering correctly. Let me investigate.

noroadsleft, 2021-03-03:
Apparently I had some cached data that was loading the previous keymap even though Configurator's data was current.

Everything should be fine now.

[image]

YangPiCui, 2021-03-04:
Wonderful. Thank you.

closed 2021-03-04 by YangPiCui