qmk/qmk_firmware #12080: [Bug] Audio Clicky is bugged.

drashna, 2021-03-02:
## Describe the Bug

On AVR, it's not too bad.  A sound get get stuck until another note is played. Not a big deal. 
On ARM, instead of getting stuck on a note, the controller locks up, and must be unplugged. 

## System Information

 - Keyboard:  
   - Revision (if applicable):  DM, Moonlander
 - Operating system: MacOS/Windows
 - AVR GCC version: 8.3.1

 - ARM GCC version: 8.4.0

 - QMK Firmware version: 0.11.x

## Additional Context

This didn't happen on develop when testing... 

I suspect that music mode and the like will trigger this too, but haven't tested. But it's pretty random when it will happen.  Sometimes it's right away, sometimes it takes half an hour.

drashna, 2021-04-18:
May be related to #8503

taotien, 2021-11-07:
Preonic rev3, seemingly only locks up when `AUDIO_ENABLE_TONE_MULTIPLEXING` is `#define`d. Otherwise, typing fast causes die.

Making with util/docker_build.sh
ARM GCC version: 6.3.1
QMK Firmware version: 0.14.29

drashna, 2021-12-24:
> Preonic rev3, seemingly only locks up when `AUDIO_ENABLE_TONE_MULTIPLEXING` is `#define`d. Otherwise, typing fast is fine for me.
> 
> Making with util/docker_build.sh ARM GCC version: 6.3.1 QMK Firmware version: 0.14.29

happens for me with or without that defined. 

Also, seems to only happen on dac basic on ARM.  DAC Additive seems to work without issue.

drashna, 2022-08-06:
@JohSchneider since you wrote the original overhaul code and have been active recently, any thoughts?

JohSchneider, 2022-08-30:
ah - a mention... sry for the delay
i vaguely remember the clicky feature as being... fineky. IIRC a "hanging note" could happen when the state for the currently playing note was not terminated properly, or too many where queuing up... which would match the observations with TONE_MULTIPLEXING.
would have to dig into the code again O:-)

JohSchneider, 2022-10-08:
> 8+ keys at once

that might be a hint - there is a stack of active notes which happens to default to 8 elements in size; do you still observe this bug when setting AUDIO_TONE_STACKSIZE to say 16?

if so, this would point to a bug in the handling of all the tones/notes that are supposed to be currently active

drashna, 2023-05-29:
> that might be a hint - there is a stack of active notes which happens to default to 8 elements in size; do you still observe this bug when setting AUDIO_TONE_STACKSIZE to say 16?
>
> if so, this would [...]

Sorry, I lost track of this.  

I'll have to double check, but the code isn't configurable, so it would be an "edit core code" test.

fsargent, 2023-05-29:
I'm having this issue on my RGBKB SOL3 when flashing from my M1 mac, so would love to try anything to fix it.

drashna, 2023-05-29:
No, changing the define to 16 doesn't help, with this, at all.  And looks like music mode can reliably trigger it on 3 simultaneous presses/notes.

fsargent, 2023-07-16:
I no longer have this issue as of updating my branch to the latest master/main branch. I think we can consider this resolved!!!

drashna, 2024-10-13:
Fixed, yeah

closed 2024-10-13 by drashna