qmk/qmk_firmware #12082: [Bug] MK_KINETIC_SPEED seems not to be fully functional

mohammedari, 2021-03-02:
I tried to enable `Kinetic Mode` feature in [mouse keys](https://docs.qmk.fm/#/feature_mouse_keys), which is recently merged to the master branch, 
however, adding `MK_KINETIC_SPEED` definition changes mouse speed oddly fast.

## Describe the Bug

As far as looking into [the implementation](https://github.com/qmk/qmk_firmware/blob/1a5f6b54aff179732e3f4f4eb79e47454f0a1eb5/quantum/mousekey.c#L111) in `mousekey.c`, the feature merged by this [PR#6739](https://github.com/qmk/qmk_firmware/pull/6739) seems to be compiled when `MK_COMBINED` is defined and `MK_KINETIC_SPEED` is **NOT** defined.

So, when adding `MK_KINETIC_SPEED` definition, **the default accelerated mode code** is executed **with kinetic mode parameter,** which is [separately defined](https://github.com/qmk/qmk_firmware/blob/master/quantum/mousekey.h#L39) in `mousekey.h`.

Is this expected behavior? Much appreciated when you kindly specify what I am missing.
Thank you.

## System Information

 - Keyboard:  Self-designed one, which is a sibling of Comet46
   - Revision (if applicable):  
 - Operating system: Ubuntu 20.04
 - AVR GCC version: avr-gcc (GCC) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) 

 - QMK Firmware version: 0.12.8

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

liyang, 2021-03-04:
Yes it's a known issue (#11766), and I'm in the process of reimplementing `MK_KINETIC_SPEED` and generally cleaning that code up so it's not as easy to make that sort of mistake again.

You can check it out at https://github.com/liyang/qmk_firmware/tree/develop/kinetic ; the behaviour is different, and it's still WIP as I intend to add jerk (i.e. quadratic rather than linear velocity). However it does give buttery-smooth movements, the documentation is almost there, and is what I'm using right now.

If you just want the existing (cleaned up and no behavioural changes) code to build, you can try https://github.com/liyang/qmk_firmware/tree/fc054c3a9caacd38519c89602308e0cdc33648ea .

Feel free to ping me on Discord if you have any questions.

mohammedari, 2021-03-05:
Thank you for pointing the related issue and PR. I got the situation.
I would wait for and am looking forward your implementation is done. Cheers!

closed 2021-03-05 by mohammedari