qmk/qmk_firmware #12155: Organize KPrepublic keyboards into the same folder

peepeetee, 2021-03-08:
Hi, I'm thinking about organizing KPrepublic keyboards into the same folder, what steps should I take?

peepeetee, 2021-03-08:
List of keyboards(not sure if comprehensive, please check):

daisy
staryu
cospad
bm16a
bm16s
bm40hsrgb
bm43a
bm60poker (manufacturer listed as KP Republic rather than KPrepublic, consider standardizing?)
bm60rgb (manufacturer listed as KP Republic rather than KPrepublic, consider standardizing?)
bm60rgb_iso
bm68rgb
dz60
jj4x4
jj40
jj50
xd002
xd004
xd60
xd68
xd75
xd84
xd87 (manufacturer listed as KP Republic rather than KPrepublic, consider standardizing?)
xd84pro
xd96

ridingqwerty, 2021-03-08:
Just reiterating some thoughts from Discord and adding some others not yet stated there:

* Pinging the "owner" of each board/keymap is considered the polite thing to do when making changes, but in this case I want to say most of these were ported in by the community rather than submitted directly by KPRepublic.  Worth checking them out, I have only spot checked a few.  If it was ported by a QMK Collaborator I would say it wouldn't really be worth pinging the submitter as we will see the PR.

* Structurally, the naive aspect of this PR is going to moving each `keyboards/<keyboard>/*` path into a vendor folder `keyboards/kprepublic/<keyboard>/*`, but may also involve updating readme info so that flashing instructions refer to this new path.

* dz60 appears to in fact be a kbdfans board, and there may be others that aren't actually KPRepublic in your list.

* I am personally unsure if it's the case that a board made by an independent creator would have their hardware sold by KPRepublic in a strict vendor relationship where KPR doesn't actually own whatever rights to the board.

* Consider submitting this to the `develop` branch.  Also note this *will* be considered a breaking change so it cannot be merged until the next breaking change submission.

closed 2021-07-08 by stale[bot]