qmk/qmk_firmware #12195: [Bug] cannot compile with keymap_german and keymap_german_osx

caplod, 2021-03-11:
I had a working config where I compile with both keymaps included, to switch between osx and windows layout

    #include "keymap_german_osx.h"
    #include "keymap_german.h"

now I get errors:

    In file included from keyboards/ergodox_ez/keymaps/caplod/keymap.c:7:0:
    quantum/keymap_extras/keymap_german.h:157:0: error: "DE_OE" redefined [-Werror]
      #define DE_OE   DE_ODIA

I think this commit changed the prefix for the osx keymap.

https://github.com/qmk/qmk_firmware/commit/f43cb314767403b8289fc957bd1d3b8250874192

but it was fixed before

https://github.com/qmk/qmk_firmware/commit/dbd4ab3457dd001ab4bbcec14315047b815f7dfe#diff-adaa707764632a0221d6f2a31ac8c222475b93f1d36b3fe82e35d2b1eaa8ee9e

Should the prefix be applied again?

fauxpark, 2021-03-11:
You can only have one of these headers included at once, since they are both German keymaps, the prefix will be `DE_` for both.

caplod, 2021-03-11:
Ok. But it was working before. I had these two includes to have osx german layout in one layer and a windows german layout in another. So I can easy use the keyboard on osx and window machines. 

wouldnt it make sense to distiguish between lets say DE_OE for windows layout and DE_OSX_OE for osx layout?

fauxpark, 2021-03-11:
The defines have been named to be at most 7 characters long for alignment purposes. It doesn't really matter anyway because these headers just contain aliases for the base keycodes - the language set in the OS is what actually determines what to print. For example `FR_A` in French AZERTY is just `KC_Q`, unless you have AZERTY set up in the OS, it's only going to print Q.

caplod, 2021-03-12:
Ok. I fixed it by having a custom keymap_german_osx.h where all defines start with DE_OSX.

closed 2021-03-12 by caplod