qmk/qmk_firmware #12222: Tracking issue for keyboards with no info.json

fauxpark, 2021-03-13:
```
2key2crawl
40percentclub/4x4
40percentclub/5x5
4pplet/aekiso60/rev_a
4pplet/steezy60/rev_a
6ball
7c8
aeboards/constellation
alpine65
aplyard/alpx6
arch_36
arisu
box75
butterstick
ck60i
four_banger
geekboards/tester
handwired/2x5keypad
handwired/412_64
handwired/42
handwired/aplx2
handwired/brain
handwired/cans12er
handwired/ck4x4
handwired/d48
handwired/dactyl_manuform/dmote/62key
handwired/daishi
handwired/datahand
handwired/hexon38
handwired/jot50
handwired/jotanck
handwired/jotpad16
handwired/k8split
handwired/myskeeb
handwired/nicekey
handwired/postageboard
handwired/riblee_f401
handwired/riblee_f411
handwired/rs60
handwired/splittest
handwired/trackpoint
handwired/traveller
hhkb_lite_2
honeycomb
ivy
kbdfans/kbdmini
keebio/viterbi
laptreus
latin47ble
latin64ble
launchpad
lets_split_eh
m10a
mechmini/v1
meira
meishi
merge/iso_macro
mschwingen/modelm
pabile/p20
pimentoso/paddino02
rgbkb/pan
runner3680
sck/gtm
splitish
standaside
technika
ungodly/launch_pad
vision_division
xelus/trinityxttkl
xelus/valor_frl_tkl
```

fauxpark, 2021-03-25:
@moyi4681 would you be able to provide an info.json for the kbdmini?

fauxpark, 2021-03-27:
`technika` appears to be [this board](https://geekhack.org/index.php?topic=104231.msg2853878#msg2853878), although the layout macro doesn't match up with the image. @Gondolindrim ?

skullydazed, 2021-06-22:
Can we remove vision_division? Its complex LAYOUT generation will never play nicely with our data-driven efforts and I don't think it's a design that more than a handful of people have, as evidence by the lack of updates over the years.

@IBNobody do you have any thoughts on this?

Gondolindrim, 2021-06-22:
> `technika` appears to be [this board](https://geekhack.org/index.php?topic=104231.msg2853878#msg2853878), although the layout macro doesn't match up with the image. @Gondolindrim ?

The reason I haven't PR'd the board to VIA and a configurator json yet is that there were changed to PCB and layout since the IC. I will PR them as soon as we have a finalized version.

closed 2021-07-23 by fauxpark