qmk/qmk_firmware #12241: [Bug] MK_COMBINED broken after PR #11804

uqs, 2021-03-14:
## Describe the Bug

1. Have #define MK_COMBINED in your config.h, as you like to use KC_ACL0 et al. when mousing
2. Build is broken:
```
Compiling: quantum/mousekey.c                                                                      quantum/mousekey.c: In function ‘move_unit’:
quantum/mousekey.c:135:35: error: ‘mouse_timer’ undeclared (first use in this function)
  135 |     } else if (mousekey_repeat && mouse_timer) {
      |                                   ^~~~~~~~~~~
quantum/mousekey.c:135:35: note: each undeclared identifier is reported only once for each function it appears in
quantum/mousekey.c: At top level:
quantum/mousekey.c:149:7: error: conflicting types for ‘mk_wheel_interval’
  149 | float mk_wheel_interval = 1000.0f / MOUSEKEY_WHEEL_INITIAL_MOVEMENTS;
      |       ^~~~~~~~~~~~~~~~~
quantum/mousekey.c:68:9: note: previous definition of ‘mk_wheel_interval’ was here
   68 | uint8_t mk_wheel_interval    = MOUSEKEY_WHEEL_INTERVAL;
      |         ^~~~~~~~~~~~~~~~~
quantum/mousekey.c: In function ‘wheel_unit’:
quantum/mousekey.c:156:35: error: ‘mouse_timer’ undeclared (first use in this function)
  156 |     } else if (mousekey_repeat && mouse_timer) {
      |                                   ^~~~~~~~~~~
 [ERRORS]
```

Patching things up to get them to compile is easy enough but results in a pointer that is _way_ too slow.

## Additional Context

Was working fine before, though I didn't actually bisect, but the PR and merge seem like obvious culprits.

zvecr, 2021-04-13:
[image]

Given the contents of #11804, I doubt that it introduced your issue.

uqs, 2021-04-13:
Sigh, sorry for the noise. Not sure what `git blame` I used to come up with this, and the good news is that this no longer reproduces,

closed 2021-04-13 by uqs