qmk/qmk_firmware #12250: error: ISO C99 requires whitespace after the macro name

H-Gerlatzek, 2021-03-15:
I just finished creating the Keymap for my first handwired build and wanted to build it using the

`make -kb <keyboard> -km <keymap>`
or
`qmk compile -kb <keyboard> -km default`

command. But when I run it it hands back following error:

`<command-line>: error: ISO C99 requires whitespace after the macro name [-Werror]`

I don't know where this error comes from or how to solve it.

fauxpark, 2021-03-15:
Your keyboard name probably has a `-` in it, change it to `_`.

H-Gerlatzek, 2021-03-15:
yes it has I'll try

H-Gerlatzek, 2021-03-15:
I created a new keyboard folder and copy + pasted my keymap and other settings.
The error seems to be solved but now I get the following error:
`error: backslash-newline at end of file [-Werror]
#define LAYOUT( \`
It also says the error is in the <keyboard>.h file.

H-Gerlatzek, 2021-03-15:
This solved the error, thank you!

closed 2021-03-15 by H-Gerlatzek

darkmatterxuan, 2024-09-23:
How?Why I can not?