qmk/qmk_firmware #12251: atusb646 and AdafruitBLE

ZJXAWC, 2021-03-15:
I have a PCB with AT90USB646 and ADAFRUITBLE. I have modified IO in the project, but Bluetooth doesn't work.
A similar design is possible on the atmega32u4. Is AT90USB646 incompatible with ADAFRUITBLE?

fauxpark, 2021-03-16:
Make sure the module has the latest firmware from Adafruit. You may also need to assign and press the `OUT_AUTO` or `OUT_BT` keycode in case the output selection isn't working.

ZJXAWC, 2021-03-16:
> Make sure the module has the latest firmware from Adafruit. You may also need to assign and press the `OUT_AUTO` or `OUT_BT` keycode in case the output selection isn't working.

I checked the cables and modules to make sure the hardware was fine.
The Bluetooth module also works well on the ATmega32U4 PCB.

AdafruitBleIRQPin B0
AdafruitBleCSPin B4
AdafruitBleResetPin B5

Are there any problems with these IO connections?

fauxpark, 2021-03-16:
Those three pins can really be anything, as long as they aren't already used. But the actual SPI pins (SCK, MOSI and MISO; ignore SS) should be according to the table here: https://docs.qmk.fm/#/spi_driver

ZJXAWC, 2021-03-17:
> Those three pins can really be anything, as long as they aren't already used. But the actual SPI pins (SCK, MOSI and MISO; ignore SS) should be according to the table here: https://docs.qmk.fm/#/spi_driver

Thank you, my problem is solved

closed 2021-03-17 by ZJXAWC