qmk/qmk_firmware #12253: [Feature Request] BM60EC RGB Support

bry95, 2021-03-15:
Hi! I'd really appreciate BM60EC support. It's nearly identical to the standard BM60 but with a rotary encoder in the top right corner. VIA firmware is on the kprepublic site if that's at all helpful. Thanks!

## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

fauxpark, 2021-03-16:
The Google Doc contains the VIA JSON, which defines the matrix layout, but it doesn't provide any information on pinout. If it's identical to the standard BM60 the matrix pins are probably the same, but the encoder pins will need to be found.

SuperVi, 2021-06-22:
Hi there, 
I recenty bought this board and would love to see it officially supported. I am quiete new to the topic but I have a background in CS and could at least test or tweak if someone with more experience simply doesn't have the board at hand. 

i found a user who forked the repo and already added the board. I think it could be useful.
https://github.com/umbacos/qmk_firmware/tree/master/keyboards/bm60ec_rgb

peepeetee, 2021-11-07:
Oh no I mentioned the wrong person, I meant @umbacos

Didel, 2022-04-17:
It looks like the solution provided in #15104 (approved by @fauxpark 8 hours ago) will be merged into QMK soon. Would that resolve this issue?

Didel, 2022-06-13:
#15104 has been approved and merged by f6a7bf2, so I assume that closes this issue.

closed 2022-10-13 by github-actions[bot]