qmk/qmk_firmware #12266: tap twice and hold to trigger scmd

vangoah20, 2021-03-17:
I set RIGHT and SCMD  in one key。So when  hold the key ，SCMD is triggered， tap twice and hold at the second ，it still be a Right sequence。I just want swap the behavor:  when hold the key ，it is a Right sequence;  ；when tap twice and hold at the second，it will be SCMD。

how could I do? which file related to this?

trudkin, 2021-04-16:
the [discord server](https://discord.gg/Uq7gcHh) might be a better place to ask. An example of the actual code would help.  While not exactly what you asked SCMD_T(kc) may be what you are looking for.

vangoah20, 2021-05-24:
> the [discord server](https://discord.gg/Uq7gcHh) might be a better place to ask. An example of the actual code would help. While not exactly what you asked SCMD_T(kc) may be what you are looking for.

Thanks. I've just set SCMD_T(KC_RIGHT). My question is not to set SCMD and RIGHT in one key. I want to change the behavor of  SCMD_T.  Now the behavor is:  if tap the key, it is right key, if hold the key, it is SCMD, if tap twice and hold, it is a sequence of right key. which I want chang to is:  if tap the key , it is right key, if hold the key, it is Sequence of right key, if tap twice and hold the key, it is SCMD.

closed 2021-09-26 by stale[bot]