qmk/qmk_firmware #12285: [Bug] Debounce on ErgoDash regressed somewhere between 0.10.35 and 0.12.18

tcrs, 2021-03-18:
## Describe the Bug

I rebased my keymap from 0.10.35 to 0.12.18, then reflashed my board. Now I get a lot of key repeats, especially on the pinkie finger keys, which makes the board almost unusable. If I rebase back onto 0.10.35 and rebuild/flash everything works fine. You can see my keymap here: https://github.com/qmk/qmk_firmware/compare/master...tcrs:ergosnap-isolike - I don't think there is anything "interesting" going on in the keymap.

## System Information

 - Keyboard:  ErgoSnap (same matrix as ErgoDash)
   - Revision (if applicable: rev1 
 - Operating system: Arch linux
 - AVR GCC version: 10.2.0

 - ARM GCC version: 10.2.0

 - QMK Firmware version: 0.12.18

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I found this issue: https://github.com/qmk/qmk_firmware/issues/6269
Which sounds similar. However the conclusion seemed to be just to fiddle with DEBOUNCE/DEBOUNCE_TYPE until stuff starts working again. I would rather understand why this has changed and how to avoid the same problem happening again in the future. `DEBOUNCE` is set to 5 in ergodash/rev1/config.h in both 0.10.35 and 0.12.18, and it doesn't look like the default algorithm has changed in common_features.mk (sym_defer_g or sym_g as it used to be named). It doesn't look like there have been any material changes in `quantum/debounce` in between 0.10.35 and 0.12.18.

Could anyone give me some tips on how to debug what is happening here? It seems like a clear regression which probably doesn't affect only ErgoDash...

balaji-ramani, 2021-03-23:
I have a DIY redox and I upgraded to 0.12.20 and saw the same issue with debounce not working.  Went back to 0.11.33 and it is working fine, with the same config.

tomboland-vocovo, 2021-03-24:
I've just built a dactyl manuform, and I was compiling the firmware from the master branch as of about a week ago.  I've been having terrible debouncing issues on my slave side.  If I build with EE_HANDS, and try both sides, then it's always the slave side which has the double-tapping issues. I'm using TRRS, and I've tried different pins on my pro-micros, and lots of different cables, doubly-triply checked all my soldering etc.  I didn't really have a baseline to go on, as it's the first keyboard I've built.  In any case, I took @makerinmaking 's comment, and checked out 0.11.33, and voila, no debouncing issues any more!

tomboland-vocovo, 2021-03-24:
I'm happy to help where I can.  I can try a git bisection perhaps.

tcrs, 2021-03-28:
I'm actually just building another keyboard. Once I've finished that I'll have a "spare" to fiddle around with and I'll try and narrow down the range of working vs. non-working versions.

BRagsdale09, 2021-04-06:
I was fighting the same issue and found this issue, thank you for that!. I've tested up to version 11.60, but 12.0 breaks it on my Kyria.
Looks to be the same issue discussed here:
https://github.com/qmk/qmk_firmware/issues/6269

yejianfengblue, 2021-05-27:
I have debounce issue on the slave side of dactyl manuform. The last two columns of slave side have chatter issue. My original problem is posted in [reddit](https://www.reddit.com/r/olkb/comments/nle6wd/handwired_dactyl_manuform_key_chatter_issue_is/).

I follow this github issue to use old version and don't set any configuration. According to my test, version 0.11.69 doesn't have chatter while version 0.12.0 has chatter. Then I test commit by commit and found the breaking commit is 1581ea48dcd48d0d3f42cc09b388c468aedec45d.

Below are my test result to version 0.12.50, no idea if it helps, just post it here.
- no any configuration, chatter
- set CONSOLE_ENABLE from no to yes in file qmk_firmware/keyboards/handwired/dactyl_manuform/5x7/rules.mk, no chatter
- set DEBOUNCE_TYPE to sym_defer_g in file qmk_firmware/keyboards/handwired/dactyl_manuform/5x7/rules.mk, no chatter
- set DEBOUNCE_TYPE to sym_eager_pr or sym_eager_pk or sym_defer_pk in file qmk_firmware/keyboards/handwired/dactyl_manuform/5x7/rules.mk, no chatter

haldiraros, 2021-06-16:
I have the same issue on Sol 2. For me it happens with basically every key on the slave side. In my case the chatter appears if I disable RGB. Same as with others 0.11.69 works fine, 0.12.0 doesn't.

Neither enabling console nor changing the debounce type change things for me. Only toggling RGB (rgb matrix) or switching to lighter animations.

I enabled `DEBUG_MATRIX_SCAN_RATE` to debug a bit. 

It seems that for me on newest QMK the chatter on keypresses on slave side happens when the matrix scan frequency goes above 900 when RGB is disabled. With simple animations that take less processing occasional double presses happen when scan frequency is around 850. With more complex ones and lower scan frequency everything works fine.

On QMK 0.11.69 matrix scan frequency reported even higher with RGB disabled and chatter does not happen.

No idea if this correlation of performance and chatter is actually important but I hope it helps

tzarc, 2021-06-16:
If you're able, can you try #11930?

haldiraros, 2021-06-17:
Tried  #11930. Chatter went away. Got 0 repeat keypresses 👍

tomboland-vocovo, 2021-06-17:
This is exciting! I'll try it tonight.

tomboland-vocovo, 2021-06-17:
This is seeming to work perfectly for me too! I'll keep an eye on this getting merged in, but I'll stick with your fork for the time being. :)

Now, if I could just get my dactyl to work when my machine wakes up, without having to unplug and re-plug it in every single time, life would be perfect!

TonehavenE, 2021-06-30:
I had similar issues, tried #11930 and it immediately went away. Nice job 👍