qmk/qmk_firmware #12308: [Bug] Multiple key stuck after flashing Preonic using preonic/rev3

pingshunhuangalex, 2021-03-20:
## Describe the Bug

Just got a brand new `DROP + OLKB PREONIC KEYBOARD MX KIT V3`. Put everything together and flash it using `QMK Toolbox 0.0.21` with `.bin` file configured using `preonic/rev3` with layout `LAYOUT_ortho_5x12`.

The flash finished successfully without any error:
```
*** STM32 DFU device connected (WinUSB): STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
*** Attempting to flash, please don't remove device
>>> dfu-util.exe -a 0 -d 0483:DF11 -s 0x08000000:leave -D "E:\Downloads\preonic_rev3_sphs_preonic_v3.bin"
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    
    Opening DFU capable USB device...
    ID 0483:df11
    Run-time device DFU version 011a
    Claiming USB DFU Interface...
    Setting Alternate Setting #0 ...
    Determining device status: state = dfuERROR, status = 10
    dfuERROR, clearing status
    Determining device status: state = dfuIDLE, status = 0
    dfuIDLE, continuing
    DFU mode device DFU version 011a
    Device returned transfer size 2048
    DfuSe interface name: "Internal Flash  "
    Downloading to address = 0x08000000, size = 55048
    
    Download	[                         ]   0%            0 bytes
    Download	[                         ]   3%         2048 bytes
    Download	[=                        ]   7%         4096 bytes
    Download	[==                       ]  11%         6144 bytes
    Download	[===                      ]  14%         8192 bytes
    Download	[====                     ]  18%        10240 bytes
    Download	[=====                    ]  22%        12288 bytes
    Download	[======                   ]  26%        14336 bytes
    Download	[=======                  ]  29%        16384 bytes
    Download	[========                 ]  33%        18432 bytes
    Download	[=========                ]  37%        20480 bytes
    Download	[==========               ]  40%        22528 bytes
    Download	[===========              ]  44%        24576 bytes
    Download	[============             ]  48%        26624 bytes
    Download	[=============            ]  52%        28672 bytes
    Download	[==============           ]  59%        32768 bytes
    Download	[===============          ]  63%        34816 bytes
    Download	[================         ]  66%        36864 bytes
    Download	[=================        ]  70%        38912 bytes
    Download	[==================       ]  74%        40960 bytes
    Download	[===================      ]  78%        43008 bytes
    Download	[====================     ]  81%        45056 bytes
    Download	[=====================    ]  85%        47104 bytes
    Download	[======================   ]  89%        49152 bytes
    Download	[=======================  ]  93%        51200 bytes
    Download	[======================== ]  96%        53248 bytes
    Download	[=========================] 100%        55048 bytes
    Download done.
    File downloaded successfully
    Transitioning to dfuMANIFEST state
*** STM32 DFU device disconnected (WinUSB): STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
*** HID console connected: OLKB Preonic (FEED:6061:0003)
```

Right after flash, the keyboard becomes unresponsive with random keycode spamming non-stop. After investigation, it looks like every time the keyboard is connected, it constantly sends signals of all the keystrokes from the first column and the 7th column. As a result, either the entire keyboard becomes unresponsive or the first and 7th column becomes unresponsive while other column working as expected (this usually happens after an OS reboot without unplugging the keyboard. Unplugging and replugging the keyboard will render the entire keyboard unresponsive again).

I've tried flashing with a default bin file, or flash it with several older version of QMK Toolbox, but no luck so far. Please any help would be greatly appreciated as the keyboard is not usable at all at the moment, nor can it be restored to a factory setting...

## System Information

 - Keyboard:  
   - Revision (if applicable):  `preonic/rev3`
 - Operating system: `Win 10`
 - AVR GCC version: Not sure, only using GUI `QMK Toolbox 0.0.21`

 - ARM GCC version: Not sure, only using GUI `QMK Toolbox 0.0.21`

 - QMK Firmware version: Not sure, only using GUI `QMK Toolbox 0.0.21`

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

pingshunhuangalex, 2021-04-09:
This is due to a faulty PCB. Everything works like a charm now with a replacement.

closed 2021-04-09 by pingshunhuangalex

ccsobral, 2021-07-19:
Hi @pingshunhuangalex , I seem to have encountered the same issue on my brand new Planck rev6 (also from Drop). Any keymap other than the eeprom reset one leads to random keys (CTRL always among them) being stuck. 
How did you pin this down to a bad PCB?

pingshunhuangalex, 2021-07-19:
I asked some friendly people in the community to test my keymap for me and it works fine in their Preonic. I also asked them to send me a keymap that works on their Preonic and no luck either. So I ran out of options besides a faulty PCB... @ccsobral

uqs, 2021-08-09:
Could you upload a known working firmware that was build with a recent QMK please?

I have to use USE_FPU=no to get things to compile under debian and none of my custom build firmwares work, they result in USB reset errors :(

ccsobral, 2021-08-18:
The only firmware I got working on my faulty Planck was this one https://discord.com/channels/440868230475677696/473506116718952450/539284620861243410

(equivalent firmware for Preonic: https://discord.com/channels/440868230475677696/473506116718952450/537849497313738763)

Hope it helps

uqs, 2021-08-18:
Yeah, those are the ones linked from the FAQ. They were build with an ancient GCC release though.

```
% strings preonic_rev3_default.bin planck_rev6_default.bin | grep GCC
GCC: 7.3.1
GCC: 7.3.1
```
I got things working with GCC 11 now by disabling both variants of the RGB support. Somehow newer GCC releases are miscompling that.

The submitter of https://github.com/qmk/qmk_firmware/issues/6753 figured this one out.