qmk/qmk_firmware #12328: [Feature Request]  Instant tap dance

LazaroFilm, 2021-03-22:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I have a Tap Dance that changes my `Shift` key into `Caps Lock` when double tapped. However, if pressed once only, the shift only starts at the end of the tapping term which is slowing me down when using shift as a modifier with the mouse in my CAD program. 
What I would like to achieve is something like this: 
```
o = key press
Time ----->
Tap dance currently:
-single tap-
o------------------------
NOTHING-----SHIFT--------
-double tap-
o-----------o------------
NOTHING-----CAPS-LOCK----

What I would like to do:
-single tap-
o------------------------
SHIFT--------------------
-double tap-
o-----------o------------
SHIFT-------CAPS-LOCK----
```
Is that achievable with QMK? If so, how can I achieve it? If not, could I request it as a feature?

EugenyPavlenko, 2021-04-03:
not sure if I can do that but I would vote for this issue to be resolved... there is no reason for the delay if the key was already released...

trudkin, 2021-04-16:
I am hardly the one to decide if this should be an enhancement.  But I believe the performance you are looking for can be achieved by using  '''get_tapping_term''' and '''get_permissive_hold'''.  Both are documented here [Tap-Hold Configuration Options](https://docs.qmk.fm/#/tap_hold?id=tap-hold-configuration-options)  With some combination of the two you should get closer to what you are looking for performance wise.

IsaacElenbaas, 2021-04-19:
The reason Mod Taps and Tap Dance do not do this currently is because of the GUI mod. Windows would open the start menu. The same behavior may occur with ALT, but I do not remember.

bodograumann, 2021-07-13:
Do I understand correctly, that this would be solved by making PERMISSIVE_HOLD have effect on tap-dances?
That would be something I'd really like to see.
I have some (mildly complex) tap/hold combinations defined with Oryx on my Ergodox EZ through tap dance, but because they are only differentiated by the TAPPING_TERM I get a lot of false positives, making the functionality unusable :cry:. (Shift-K / Shift; Shift-J / Shift; Ctrl-J / Ctrl; Ctrl-6 / Super; Ctrl-W / MO(1); Esc / MO(1); Meta+H / Super; Meta+L / Ctrl)

IsaacElenbaas, 2021-07-14:
Just a bit of trivia, `PERMISSIVE_HOLD` actually doesn't change the modifier activation time, see [here.](https://cdn.discordapp.com/attachments/451392568345755648/747388373781708930/tap-holds-qmk.pdf) Why, I have no clue - my pull hooking into that section of the code for Retro Shift registers immediately with no issues. But yes, it should, and Tap Dance could probably register sooner too if it respected `PERMISSIVE_HOLD`, though I'm not familiar with its code.

What @LazaroFilm describes, at least, can easily be accomplished with [custom Tap Dance functions.](https://beta.docs.qmk.fm/using-qmk/software-features/feature_tap_dance#complex-examples-id-complex-examples) Ideally Tap Dance would change its key registration time based on whether there even is a single/double tap action it could use instead, but that would be ridiculously complicated.

Tap Dance does respect `TAPPING_TERM_PER_KEY`, by the way, which could help you cut down on those false positives.

closed 2022-01-09 by stale[bot]