qmk/qmk_firmware #12335: Keyboard sirius/uni660/rev2/ansi compile error

hades2313, 2021-03-22:
I tried to compile keyboard sirius/uni660/rev2/ansi on our qmk configurator UI + my windows machine + my mac machine and all received an this error:

```
Linking: .build/sirius_uni660_rev2_ansi_layout_ansi_mine.elf                                        
 | 
 | .build/obj_sirius_uni660_rev2_ansi_layout_ansi_mine/matrix.o: In function `matrix_init':
 | /qmk_compiler/qmk_firmware/keyboards/sirius/uni660/rev2/matrix.c:134: undefined reference to `serial_init'
 | collect2: error: ld returned 1 exit status
 | 
```

I've noticed that the header file containing serial_init() function is included in matrix.c, so I don't know why we have undefined reference here without a deep dive. I've also contacted the author of this keyboard and he is not sure as well and he said it was working last year. Maybe it's related with the way "qmk compile" command works and it has changed since last year?

Look forward to hearing from you, thank you for your help!

noroadsleft, 2021-03-22:
Seems to have been caused by #11521. @fauxpark any ideas?

fauxpark, 2021-03-23:
Whoops, I must have forgotten to add serial_uart.c to SRC. PR incoming.

PR #12341 (2021-03-24): Fix Uni660 linker errors

## Description

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #12335 

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have made the appropriate ch [... cut]
closed 2021-03-24 by fauxpark