qmk/qmk_firmware #12339: YMDK 68 Keyboard Support?

molobito, 2021-03-22:
Hi, could not find the YMDK 68 keyboard in QMK keyboard list... referenced as YM68, YMD68 and MT68. 

I got the keyboard from Aliexpress:
https://www.aliexpress.com/item/32999629504.html

I got this JSON file off the google drive links in Aliexpress:
https://drive.google.com/file/d/1NdVerLbV7KVJ9B3AmdrgIasEoMhMIgpP/view
[ym68.json.gz](https://github.com/qmk/qmk.fm/files/6177959/ym68.json.gz)

Inside the JSON file the name is YM68 but the kbname says MT68 but couldn't find either in QMK keyboards

I was able to download older QMK source code files from a chinese QMK firmware builder site (http://mtkeyboard.vip/) but couldn't make them work =(  The keyboard there was listed as YM

[ymd68.zip](https://github.com/qmk/qmk.fm/files/6177933/ymd68.zip)

noroadsleft, 2021-03-22:
So apparently I have a branch with support for this board, but I never got around to uploading it.

Give me maybe a day to PR it; I'm busy with some other stuff.

molobito, 2021-03-23:
Fantastic!  No rush...and thanks in advance!

PR #12352 (2021-03-24): [Keyboard] YMDK YM68
## Description

Converted from a KBFirmware JSON file provided via the vendor's product listing.

PCB supports Backlight on B6 and RGB Underglow on E2, but the LEDs are not pre-soldered according to the PCB photos.

## Types of Changes

- [x] Keyboard (addition or update)

## Issues Fixed or Closed by This PR

* Closes #12339.

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk [... cut]
closed 2021-03-24 by noroadsleft