qmk/qmk_firmware #12368: [Bug] QMK compile hangs immediately with kbdpad_mk2

Nebrethar, 2021-03-24:
## Describe the Bug

I am trying my first compile and the program seems to get stuck waiting on a python subprocess, right when it starts.

If I stop the execution immediately after starting to compile, I get the same traceback as if I waited 30 minutes for it to run. My WSL is kind of slow, but not that slow 😁 

Please let me know if I am posting the issue in the wrong place.

## System Information

 - Keyboard:  kbdfans/kbdpad_mk2
 - Operating system: WSL 2 with Ubuntu 20.04
 - AVR GCC version: 5.4.0
 - ARM GCC version: 9.2.1 20191025
 - QMK Firmware version: 0.12.20
 - Any keyboard related software installed? 
   - QMK Toolbox (On the windows side)

## Additional Context

Traceback:
```
qmk compile -kb kbdfans/kbdpad_mk2/ -km default
Ψ Compiling keymap with make -j 1 kbdfans/kbdpad_mk2/:default

QMK Firmware 0.12.20
^C<class 'KeyboardInterrupt'>
Traceback (most recent call last):
  File "/home/nebs/.local/lib/python3.8/site-packages/milc/milc.py", line 465, in __call__
    return self.__call__()
  File "/home/nebs/.local/lib/python3.8/site-packages/milc/milc.py", line 470, in __call__
    return self._entrypoint(self)
  File "/mnt/c/Users/nebrethar/Documents/GitHub/qmk_firmware/lib/python/qmk/decorators.py", line 42, in wrapper
    return func(*args, **kwargs)
  File "/mnt/c/Users/nebrethar/Documents/GitHub/qmk_firmware/lib/python/qmk/decorators.py", line 85, in wrapper
    return func(*args, **kwargs)
  File "/mnt/c/Users/nebrethar/Documents/GitHub/qmk_firmware/lib/python/qmk/cli/compile.py", line 68, in compile
    compile = cli.run(command, capture_output=False, text=False)
  File "/home/nebs/.local/lib/python3.8/site-packages/milc/milc.py", line 146, in run
    return subprocess.run(command, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 491, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1016, in communicate
    self.wait()
  File "/usr/lib/python3.8/subprocess.py", line 1079, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1804, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.8/subprocess.py", line 1762, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt
```

tzarc, 2021-03-24:
Does it work if you remove the trailing slash on the keyboard?

You've got:
```
qmk compile -kb kbdfans/kbdpad_mk2/ -km default
```
...but it should be:
```
qmk compile -kb kbdfans/kbdpad_mk2 -km default
```

Nebrethar, 2021-03-24:
@tzarc Thanks for the reply. Removing the trailing slash also hangs (with the same traceback)

tzarc, 2021-03-24:
Does this happen with other keyboards? Try something like:

```
qmk compile -kb planck/rev6 -km default
```

Nebrethar, 2021-03-24:
The program was also hanging with this build.. but I have another update!!!

I had the window for my first attempt at compiling for kbdpad_mk2 on a separate monitor just in case it just took a long time. About 3 hours later, it finally started! I am here at my computer and it started going all of a sudden. Went quickly from there. Weird!

tzarc, 2021-03-24:
Very!

You can give this a try, too -- it'll attempt to skip a bunch of the pre-work:
```
qmk compile -kb kbdfans/kbdpad_mk2 -km default -e SKIP_GIT=yes -e SKIP_VERSION=yes
```

Nebrethar, 2021-03-24:
Thanks @tzarc for your help on this thus far. I have my compiled code but this is a puzzle I want to solve now! 😁 

I'll have to pick this up again tomorrow. Let me know if you have any ideas on how I could get a very verbose reading from these subprocesses...

tzarc, 2021-03-24:
For that....we can add a `VERBOSE` flag:
```
qmk compile -kb kbdfans/kbdpad_mk2 -km default -e SKIP_GIT=yes -e SKIP_VERSION=yes -e VERBOSE=1
```

Nebrethar, 2021-03-25:
I ran the above command and waited a while. I don't get any output at all for just as long, and the program seems to work quickly and successfully once it starts up.

This is the first line I see after a couple hours, followed by the rest of the verbose output:
```
# Check if we have the CMP tool installed
```

It looks like some python subprocess is waiting and looping. I am going to try on an Ubuntu (not WSL) PC to narrow it down.

fauxpark, 2021-03-25:
```
/mnt/c
```
Here is your problem. IO is extremely bad in WSL2, for some reason, when accessing the host filesystem. Move your repo to the WSL filesystem (`\\wsl$`) and you should see a significant speed increase.

Nebrethar, 2021-03-26:
Thanks for the help on this.

closed 2021-03-26 by Nebrethar