qmk/qmk_firmware #12370: [Request for comments] Several build targets for the IBM Model M are cluttered in different folders

nuess0r, 2021-03-24:
In QMK are already three different build targets that handle the IBM model M. 
All three are in different folders and different folder structures. None of them is listed in the community supported keyboards.

I'd like to discuss the way forward here before making any changes. Hopefully I reach all existing maintainers.

## Describe the Bug

The IBM Model M is popular among keyboard enthusiasts and it's not a surprise that more than one hardware project exists to replace the original controller from the 80's.

Currently there are three in QMK:
  - /keyboards/mschwingen/modelm/ by @mschwingen 
  - /keyboards/yugo_m/ by @tomic1785
  - /keyboards/converter/modelm101/ by @iw0rm3r 

And two upcoming ones:
  - https://github.com/jhawthorn/qmk_firmware/tree/modelh_stm32/keyboards/modelh by @jhawthorn 
  - https://github.com/nuess0r/ctrl-M by myself

There are at least this issues I'd like to raise:
  - It is obvious that QMK is undecided if replacement controllers shall be keyboards or converters
  - This structure does not allow to share common code between the targets as it is done for other keyboards.
  - Also it makes it hard for a user to know what support for the Model M exists.

My initial proposal would be to:
  - create /keyboards/modelm
  - move the three existing projects below this folder
  - write a common readme.md describing the Model M and the different variants an user could buy/build

I'm very happy for your comments and discussions. I think we (all Model M lovers) can profit from it.

## System Information

 - Keyboard:  IBM Model M

## Additional Context

I guess that when we better organize this separate controller projects targeting a keyboard that always looks the same, it would be possible to share user keymaps between different controllers (correct me if there is a QMK limitation against this).

tomic1785, 2021-03-25:
I think that putting all the projects under the same directory might be a good idea in the sense of making the huge QMK keyboard repo easier to navigate. In terms of marketing and selling hardware (if that's your aim) it wouldn't make much difference, but there is no reason not to proceed with it. 

Now on the topic of having multiple projects that serve the same purpose: 
When I was developing my own solution, there were no projects around that even closely replicated the funcionality of the original IBM's PCB (plus USB and customizability) for a reasonable price. That, in fact, was the reason I made this controller. 
There were (are?) two more QMK projects, I don't remember the names; one was with AT90USB1286 MCU (that was enough of a drawback itself at the time when STM32 was 2-3$), and the other was with ATMEGA32U4 and also didn't fit my needs. 
Yugo-M sure has some shortcomings (no cover around the USB plug and locklight cable with different connectors), that I decided not to change for the sake of keeping it economical and easy to build. 
And then we come to the main problem (for me at least): sourcing parts. No matter if you make the PCBs or the users make them themselves, inability to buy a part automatically puts the project off the table. My particular project, although completely finished (for now at least), isn't published or available for purchase, because of the situation where it's pretty much impossible to find a remotely reliable source of the STM32F303 chips. Unfortunately this is going on for quite some time. 

And that's why it's important to have some diversity here and develop the same thing on different platforms. 

**Tl;dr**: 
Yeah, it's a good idea, why not. Also we shold continue the development each in their own direction.

mschwingen, 2021-03-25:
On 24.03.21 22:49, nuess0r wrote:
> In QMK are already three different build targets that handle the IBM
> model M.
> All three are in different folders and different folder structures.
> None of them is listed in the community supported [...]

IMHO, a converter is something external that is attached between a
keyboard and a PC, so I consider the Model M PCBs to be a "keyboard" and
not a "converter", but if people think otherwise, I can live with that.

> * This structure does not allow to share common code between the
>     targets as it is done for other keyboards.
>   * Also it makes it hard for a user to know what support for the
>     Model M exists. [...]
I have no objections against moving the designs into a common directory.

I am not sure if much code can be shared: the hardware is quite
different, even if all designs target the same matrix. Note, for
example, that I had to rotate the matrix due to the use of 74HC165 as
input extenders, making the "row" pins inputs, and LED handling may also
depend on the possibilities of the design, but it does not hurt to try.

cu

Michael

nuess0r, 2021-03-25:
Thanks for your fast reply @tomic1785 and @mschwingen. I hope some of the QMK maintainers will also reply.

Now reading again my post, I realized that I forgot the forth Model M controller in the QMK repo:
/keyboards/ashpil/modelm_usbc/ by @ashpil 

I agree that it is OK to have several different Model M controllers. If it is for different generations of microcontrolers (There will be at some point a RiscV Model M controller :-) )There are also many variants of original IBM controllers with different connectors and locations. Another one I found on ebay includes LiPo charger and Bluetooth. It seems to be running with QMK but I didn't find the design of it.

I would be happy if we could share some of the work needed on the documentation. All of us have to provide some information on how to replace the controller, check which Model M variant is at hand (gather some pictures) and what controller would fit into it, additional mods etc.

ashpil, 2021-03-25:
>  /keyboards/ashpil/modelm_usbc/ by @ashpil

Wonderful to be involved! I hadn't realized so many more alt controllers have popped up since I created mine. 

> It is obvious that QMK is undecided if replacement controllers shall be keyboards or converters

I remember this [discussion](https://github.com/qmk/qmk_firmware/pull/7393#issuecomment-554836945) when I submitted the PR for my controller. There was the idea of putting it in `/converters`, but at the time I argued against it. I'm not sure if I still feel that way - it sort of is a converter from the old Model M to a new version, but I guess that depends on the definition of "converter" you use.

> This structure does not allow to share common code between the targets as it is done for other keyboards.

I echo @mschwingen's concern here - I'm not sure if there is all that much that could be shared as common code. Can you give some examples of common code that is shared for other keyboards?

> create /keyboards/modelm

Sounds good to me.

> All of us have to provide some information on how to replace the controller, check which Model M variant is at hand (gather some pictures) and what controller would fit into it, additional mods etc.

It would certainly be a good thing to have this information somewhere, but I'm not sure QMK is the place for it - does it have this sort of guide-level documentation for other keyboards?

> it would be possible to share user keymaps between different controllers 

This would be nice.

nuess0r, 2021-03-27:
>>This structure does not allow to share common code between the targets as it is done for other keyboards.

>I echo @mschwingen's concern here - I'm not sure if there is all that much that could be shared as common code. Can you give some examples of common code that is shared for other keyboards?

>> it would be possible to share user keymaps between different controllers

> This would be nice.

I can't give examples yet as I have not started digging into the QMK code. Also it's something optional/mid term goal and I added it as idea and/or argument for you other contributors to be more interested. But you were all convinced very fast :-)

If the only thing we share in the end are keymaps than I'm already quite happy with the outcome of this effort.

>> All of us have to provide some information on how to replace the controller, check which Model M variant is at hand (gather some pictures) and what controller would fit into it, additional mods etc.

> It would certainly be a good thing to have this information somewhere, but I'm not sure QMK is the place for it - does it have this sort of guide-level documentation for other keyboards?

The guides/documentation go to qmk.fm. It's not used yet very much for most keyboards but some of them have nicer documentation there: https://qmk.fm/keyboards/vitamins_included/

rljacobson, 2023-05-26:
Cross reference PR #12821, a port of TMK's ibmpc_usb converter code. That PR was closed by the author without explanation. It appears to me to be complete, but I haven't tried it out. It's a converter, not a controller.

nuess0r, 2023-05-30:
Finally it's done!
https://github.com/qmk/qmk_firmware/blob/master/docs/ChangeLog/20230528.md

All controller projects supporting a IBM Model M variant are now under one common folder keyboards/ibm/:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/ibm

closed 2023-05-30 by nuess0r