qmk/qmk_firmware #12421: Make fails with "ERROR: Can not run bin/qmk!"

Gr1mR3aver, 2021-03-29:
make is unable to run.  This appears to be due to qmk being unable to find a python module, even though it is installed.  running `qmk` returns an error `Could not find module pillow!` even though the module is installed.

## Describe the Bug

running utils/qmk_install.sh shows the module is installed (paths removed):
`Requirement already satisfied: appdirs in [...] (1.4.4)
Requirement already satisfied: argcomplete in [...] (1.12.2)
Requirement already satisfied: colorama in [...] (0.4.3)
Requirement already satisfied: dotty-dict in [...] (1.3.0)
Requirement already satisfied: hjson in [...] (3.0.2)
Requirement already satisfied: jsonschema>=3 in [...] (3.2.0)
Requirement already satisfied: milc>=1.1.0 in [...]  (1.3.0)
Requirement already satisfied: pygments in [...] (2.7.3)
Requirement already satisfied: pillow in [...] (8.1.0)
Requirement already satisfied: setuptools-scm in [...] (5.0.1)
Requirement already satisfied: setuptools in [...] (45.2.0)`

but when running `qmk`, the first line displayed is:
`Could not find module pillow!
`
This breaks make.
## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: Windows w/WSL2
 - AVR GCC version: 5.4.0

 - ARM GCC version: (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 - QMK Firmware version: 0.12.15

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Using VSCode on windows 10 running qmk in a wsl2 terminal

fauxpark, 2021-03-29:
Run `python3 -m pip install -U qmk`

Gr1mR3aver, 2021-03-29:
that completed with no errors but did not fix the issue.  interestingly, pillow was not in the list of checked packages.  Output below, with paths obfuscated:

`Requirement already up-to-date: qmk in [homedir]/.local/lib/python3.8/site-packages (0.0.45)
Requirement already satisfied, skipping upgrade: colorama in /usr/lib/python3/dist-packages (from qmk) (0.4.3)
Requirement already satisfied, skipping upgrade: flake8 in [homedir]/.local/lib/python3.8/site-packages (from qmk) (3.8.4)
Requirement already satisfied, skipping upgrade: argcomplete in [homedir]/.local/lib/python3.8/site-packages (from qmk) (1.12.2)
Requirement already satisfied, skipping upgrade: nose2 in [homedir]/.local/lib/python3.8/site-packages (from qmk) (0.9.2)
Requirement already satisfied, skipping upgrade: milc>=1.3.0 in [homedir]/.local/lib/python3.8/site-packages (from qmk) (1.3.0)
Requirement already satisfied, skipping upgrade: appdirs in [homedir]/.local/lib/python3.8/site-packages (from qmk) (1.4.4)
Requirement already satisfied, skipping upgrade: pygments in [homedir]/.local/lib/python3.8/site-packages (from qmk) (2.7.3)
Requirement already satisfied, skipping upgrade: yapf in [homedir]/.local/lib/python3.8/site-packages (from qmk) (0.30.0)
Requirement already satisfied, skipping upgrade: hjson in [homedir]/.local/lib/python3.8/site-packages (from qmk) (3.0.2)
Requirement already satisfied, skipping upgrade: dotty-dict in [homedir]/.local/lib/python3.8/site-packages (from qmk) (1.3.0)
Requirement already satisfied, skipping upgrade: jsonschema>=3 in /usr/lib/python3/dist-packages (from qmk) (3.2.0)
Requirement already satisfied, skipping upgrade: pycodestyle<2.7.0,>=2.6.0a1 in [homedir]/.local/lib/python3.8/site-packages (from flake8->qmk) (2.6.0)
Requirement already satisfied, skipping upgrade: pyflakes<2.3.0,>=2.2.0 in [homedir]/.local/lib/python3.8/site-packages (from flake8->qmk) (2.2.0)
Requirement already satisfied, skipping upgrade: mccabe<0.7.0,>=0.6.0 in [homedir]/.local/lib/python3.8/site-packages (from flake8->qmk) (0.6.1)
Requirement already satisfied, skipping upgrade: coverage>=4.4.1 in [homedir]/.local/lib/python3.8/site-packages (from nose2->qmk) (5.3.1)
Requirement already satisfied, skipping upgrade: six>=1.7 in /usr/lib/python3/dist-packages (from nose2->qmk) (1.14.0)
Requirement already satisfied, skipping upgrade: setuptools-scm in [homedir]/.local/lib/python3.8/site-packages (from dotty-dict->qmk) (5.0.1)
Requirement already satisfied, skipping upgrade: setuptools in /usr/lib/python3/dist-packages (from setuptools-scm->dotty-dict->qmk) (45.2.0)`

Gr1mR3aver, 2021-03-29:
might be worth noting that this was working until I switched branches.

fauxpark, 2021-03-29:
What branch?

fauxpark, 2021-03-29:
And yes, `pillow` is not currently required by `qmk` or any of its dependencies.

Gr1mR3aver, 2021-03-29:
Dupe of 12413

closed 2021-03-29 by Gr1mR3aver