qmk/qmk_firmware #12446: [Feature Request] Command to edit info.json files

skullydazed, 2021-03-31:
## Feature Request Type

- [x] Core functionality
- [x] New behavior

## Description

We now have the pieces in place to build a TUI for editing a keyboard. The `info.json` can be used to describe all the basic attributes of a keyboard and we have a jsonschema for that file in [`data/schemas/keyboard.jsonschema`](https://github.com/qmk/qmk_firmware/blob/master/data/schemas/keyboard.jsonschema). 

We should have a `qmk edit -kb <keyboard>` command that gives a friendly interface for editing. It should use the schema file to know what a valid `info.json` looks like, and when new items are added to the schema it should not be necessary to change `qmk edit`. The interface should be similar to other configuration TUI's, such as raspberry pi's `raspi-config` or `make menuconfig` used by linux, busybox, and other projects.

When it comes to choosing a TUI library for this long term maintainability is the most important factor. It would be nice if we could use the curses library and not have to include more dependencies, but that library is pretty low level and doesn't work on windows. I'll leave the selection up to whoever is going to implement this, but if you want to use something niche please talk to us before getting started. Some interesting possibilities to consider:

* http://urwid.org/
* https://pypi.org/project/blessed/
* https://github.com/jwlodek/py_cui
* https://github.com/peterbrittain/asciimatics

Lockna, 2021-09-02:
Hey, I would like to take over this issue and implement the feature! ^^

TweetyDaBird, 2022-09-30:
If this is still desirable, after banging my head at doing .json files for a few days, I think I'd be motivated to tackle this.

I sincerely doubt it will be completely possible to not need to edit what I come up with for new options, but to some extent it should work reading from the json schema.

Anika-Roy, 2022-10-11:
Hi! Im participating in Hacktoberfest and I'd like to know if this issue is counted in Hacktoberfest and if I can work on it?

2ynn, 2022-11-09:
Hi @skullydazed, 

I've started experimenting with urwid (more specifically `TreeWalker` and `TreeListBox` classes) and I would be interested in tackling this issue. 

I see one blocker though: as soon as I call urwid from within your cli environment (i.e. `$ qmk --color edit-info-json  -kb <keyboard>` ) I loose all of urwid's palette functionalities (colors, standout, underline).  For instance you can try and run the [palette example](https://github.com/urwid/urwid/blob/master/examples/palette_test.py) from within a `milc` subcommand and it should render as Monochrome (tested on `urxvt` and `xterm`).

It looks like `milc` renders ANSI colors in its own way and I doubt using another tui library would help. I though about starting a subprocess in another terminal but I believe I would still need to know which terminal the user is using and it kind of defeats the purpose of using `milc` (especially in regards to compatibility with windows). Any thoughts?

Thanks!

stuartsoft, 2023-02-22:
@2ynn did you make progress on this?

2ynn, 2023-02-22:
> @2ynn did you make progress on this?

no haven't progressed on that. Have you reproduced the issue i'm talking about?

Mohabz-911, 2024-01-24:
Hello @2ynn , I would like to work on this feature. I don't understand the problem you are facing. Can you provide me with more details?