qmk/qmk_firmware #12455: qmk setup command fails?

Oi27, 2021-04-01:
While setting up the QMK environment with Msys2 on Windows, things installed correctly until the qmk setup command. It throws the error below. What does an import error mean and how can I fix it?
`$ qmk setup
Error: %s: %s ('ImportError', ImportError("cannot import name 'format_ansi' from 'milc' (D:/oi23/documents/qmk_toolbox/msys2/mingw64/lib/python3.8/site-packages/milc/__init__.py)"))
Traceback (most recent call last):
  File "D:/oi23/documents/qmk_toolbox/msys2/mingw64/lib/python3.8/site-packages/qmk_cli/script_qmk.py", line 99, in main
    import qmk.cli  # noqa
  File "D:/oi23/Documents/QMK_Toolbox/qmk_firmware/lib/python/qmk/cli/__init__.py", line 13, in <module>
    from . import doctor
  File "D:/oi23/Documents/QMK_Toolbox/qmk_firmware/lib/python/qmk/cli/doctor.py", line 12, in <module>
    from qmk.questions import yesno
  File "D:/oi23/Documents/QMK_Toolbox/qmk_firmware/lib/python/qmk/questions.py", line 4, in <module>
    from milc import cli, format_ansi
ImportError: cannot import name 'format_ansi' from 'milc' (D:/oi23/documents/qmk_toolbox/msys2/mingw64/lib/python3.8/site-packages/milc/__init__.py)
`

fauxpark, 2021-04-01:
Make sure your repo is up to date. `qmk setup` is calling some CLI code in `qmk_firmware` which tries to import `format_ansi` from MILC, but that function doesn't exist anymore.

Oi27, 2021-04-01:
> 
> 
> Make sure your repo is up to date. `qmk setup` is calling some CLI code in `qmk_firmware` which tries to import `format_ansi` from MILC, but that function doesn't exist anymore.

Updating it worked partially - now it begins to install, says there are missing packages, and asks to install them. Saying yes seems to freeze the program - not that it doesn't respond, but it doesn't do anything and says it's working on stuff when you go to close it. Here's the output below:
$ qmk setup
INFO Found qmk_firmware at D:/oi23/Documents/QMK_Toolbox/qmk_firmware.
INFO QMK Doctor is checking your environment.
INFO Detected Windows.
INFO QMK home: D:/oi23/Documents/QMK_Toolbox/qmk_firmware
WARNING QMK home does not appear to be a Git repository! (no .git folder)
ERROR Can't find arm-none-eabi-gcc in your path.
ERROR Can't find avr-gcc in your path.
ERROR Can't find avrdude in your path.
ERROR Can't find dfu-programmer in your path.
ERROR Can't find dfu-util in your path.

Would you like to install dependencies? [Y/n] Y

fauxpark, 2021-04-02:
```
WARNING QMK home does not appear to be a Git repository! (no .git folder)
```
Indicates you downloaded the .zip file from GitHub, rather than letting `qmk setup` clone the repository. MAke sure everything you want to keep in there is backed up, then delete the `qmk_firmware` folder and run `qmk setup` again.

Oi27, 2021-04-03:
> Make sure everything you want to keep in there is backed up, then delete the `qmk_firmware` folder and run `qmk setup` again.

Allowing it to set up its own repository fixed that error, but the dependencies are having trouble installing themselves. It asks if I'd like to, yes, and then it hangs up. It doesn't throw an error or anything; it just sits doing nothing until you close the window. Here's the output:
$ qmk setup
INFO Found qmk_firmware at C:/Users/oi23/qmk_firmware.
INFO QMK Doctor is checking your environment.
INFO Detected Windows.
INFO QMK home: C:/Users/oi23/qmk_firmware
ERROR Can't find arm-none-eabi-gcc in your path.
ERROR Can't find avr-gcc in your path.
ERROR Can't find avrdude in your path.
ERROR Can't find dfu-programmer in your path.
ERROR Can't find dfu-util in your path.

Would you like to install the dependencies? [Y/n] Y

fauxpark, 2021-04-04:
This line:
```
INFO Detected Windows.
```
should be a little more detailed now, so your repo is not quite up to date.

`python3 -m pip install -U qmk` will also update the CLI Python package if it is outdated.

Oi27, 2021-04-04:
Updating the CLI Python package didn't change anything, so I did a fresh install and QMK is up and running after putting the MingW64.exe into the QMK_MSYS folder. Thank you for the help!

closed 2021-04-04 by Oi27

cartazio, 2021-04-18:
i'm seeing the same issue on OSX with the brew install method, my current theory is that somehow the brew / pip based install paths on mac have some env leakage, so i'm trying out the nix shell on my computer to see if that works

cartazio, 2021-04-18:
```
 $ /usr/local/Cellar/qmk/0.0.45/bin/qmk
Error: %s: %s ('ImportError', ImportError("cannot import name 'format_ansi' from 'milc' (/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/milc/__init__.py)"))
Traceback (most recent call last):
  File "/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/qmk_cli/script_qmk.py", line 99, in main
    import qmk.cli  # noqa
  File "/Users/carter/Desktop/repoScratcher/qmk_firmware/lib/python/qmk/cli/__init__.py", line 13, in <module>
    from . import doctor
  File "/Users/carter/Desktop/repoScratcher/qmk_firmware/lib/python/qmk/cli/doctor.py", line 13, in <module>
    from qmk.questions import yesno
  File "/Users/carter/Desktop/repoScratcher/qmk_firmware/lib/python/qmk/questions.py", line 4, in <module>
    from milc import cli, format_ansi
ImportError: cannot import name 'format_ansi' from 'milc' (/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/milc/__init__.py)
```

fauxpark, 2021-04-19:
@cartazio your repo is out of date, as mentioned earlier. The Python code in the qmk_firmware repo is expecting one of the dependencies (MILC) of the QMK CLI Python *package* to have a function named `format_ansi()`, however this has since been refactored and neither MILC nor qmk_firmware refer to it anymore. I can see your CLI version is 0.0.45 which is the latest as of this comment, so the issue is on the qmk_firmware end.

If you have not committed to your master branch you can simply run `git pull` from the qmk_firmware directory.

cartazio, 2021-04-19:
Yup I realized that later today and got stuff mostly working. ;) 
My confusion was due to a friend wanting me to use his qmk branch for some patches and thus the discrepancy hit me

OBenjaminT, 2021-06-01:
I have the same problem in WSL. The qmk cli was installed just before. It's also the thing cloning the repo.

```
➜ qmk setup
☒ Could not find qmk_firmware!
Would you like to clone qmk/qmk_firmware to /home/u/qmk_firmware? [y/n] y
Cloning into '/home/u/qmk_firmware'...
Checking out files:  82% (20087/24450)
Checking out files:  83% (20294/24450)
Checking out files:  84% (20538/24450)
Checking out files:  85% (20783/24450)
Checking out files:  86% (21027/24450)
Checking out files:  87% (21272/24450)
Checking out files:  88% (21516/24450)
Checking out files:  89% (21761/24450)
Checking out files:  90% (22005/24450)
Checking out files:  91% (22250/24450)
Checking out files:  92% (22494/24450)
Checking out files:  93% (22739/24450)
Checking out files:  94% (22983/24450)
Checking out files:  95% (23228/24450)
Checking out files:  96% (23472/24450)
Checking out files:  97% (23717/24450)
Checking out files:  98% (23961/24450)
Checking out files:  99% (24206/24450)
Checking out files: 100% (24450/24450)
Checking out files: 100% (24450/24450), done.
Submodule 'lib/chibios' (https://github.com/qmk/ChibiOS) registered for path 'lib/chibios'
Submodule 'lib/chibios-contrib' (https://github.com/qmk/ChibiOS-Contrib) registered for path 'lib/chibios-contrib'
Submodule 'lib/googletest' (https://github.com/qmk/googletest) registered for path 'lib/googletest'
Submodule 'lib/lufa' (https://github.com/qmk/lufa) registered for path 'lib/lufa'
Submodule 'lib/printf' (https://github.com/qmk/printf) registered for path 'lib/printf'
Submodule 'lib/ugfx' (https://github.com/qmk/uGFX) registered for path 'lib/ugfx'
Submodule 'lib/vusb' (https://github.com/qmk/v-usb) registered for path 'lib/vusb'
Cloning into '/home/u/qmk_firmware/lib/chibios'...
Cloning into '/home/u/qmk_firmware/lib/vusb'...
Cloning into '/home/u/qmk_firmware/lib/printf'...
Cloning into '/home/u/qmk_firmware/lib/chibios-contrib'...
Cloning into '/home/u/qmk_firmware/lib/googletest'...
Cloning into '/home/u/qmk_firmware/lib/ugfx'...
Cloning into '/home/u/qmk_firmware/lib/lufa'...
Submodule path 'lib/chibios': checked out 'ffe54d63cb10a355add318f8e922e39f1c3d4bfd'
Submodule path 'lib/chibios-contrib': checked out '61baa6b036138c155f7cfc5646d833d9423f3243'
Submodule path 'lib/googletest': checked out 'ec44c6c1675c25b9827aacd08c02433cccde7780'
Submodule path 'lib/lufa': checked out 'ce10f7642b0459e409839b23cc91498945119b4d'
Submodule path 'lib/printf': checked out 'd3b984684bb8a8bdc48cc7a1abecb93ce59bbe3e'
Submodule path 'lib/ugfx': checked out '40b48f470addad6a4fb1177de1a69a181158739b'
Submodule path 'lib/vusb': checked out 'bdb53e4c043d089279d9891b68bea77614cb97ee'
Ψ Successfully cloned https://github.com/qmk/qmk_firmware to /home/u/qmk_firmware!
Ψ Added https://github.com/qmk/qmk_firmware as remote upstream.
Error: %s: %s ('ImportError', ImportError('Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll'))
Traceback (most recent call last):
  File "/home/u/.local/lib/python3.7/site-packages/qmk_cli/script_qmk.py", line 76, in main
    import qmk.cli  # noqa
  File "/home/u/qmk_firmware/lib/python/qmk/cli/__init__.py", line 158, in <module>
    from . import console  # noqa
  File "/home/u/qmk_firmware/lib/python/qmk/cli/console.py", line 9, in <module>
    import hid
  File "/home/u/.local/lib/python3.7/site-packages/hid/__init__.py", line 30, in <module>
    raise ImportError(error)
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll
```

fauxpark, 2021-06-02:
@OBenjaminT this is a different problem, you need to install hidapi.

OBenjaminT, 2021-06-02:
Thanks, I did that, then had some other problems but existing issues already answered those.

RohitWaghmar, 2023-09-23:
Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
☒ Can't find arm-none-eabi-gcc in your path.
☒ Can't find avr-gcc in your path.
☒ Can't find avrdude in your path.
☒ Can't find dfu-programmer in your path.
☒ Can't find dfu-util in your path.

I have this problem i need help to solove this

miguelgargallo, 2024-01-12:
> i need help to solove this

me too

keyboard-magpie, 2024-01-12:
Replying to a closed issue is not the way to seek help. 

It would be better to come and ask questions on the [QMK discord server](https://discord.gg/qmk), or post a new issue with your specific problem.

miguelgargallo, 2024-01-12:
> Replying to a closed issue is not the way to seek help.
> 
> It would be better to come and ask questions on the [QMK discord server](https://discord.gg/qmk), or post a new issue with your specific problem.

Thanks for the advice, I would do that, my apologies!