qmk/qmk_firmware #12465: [Bug] qmk doctor fails in Linux/WSL

washingtonsteven, 2021-04-02:
## Describe the Bug

Running `qmk setup` (which runs `doctor`) in WSL (also probably happens in Linux Proper) produced this error:

```sh
<class 'AttributeError'>
☒ 'PosixPath' object has no attribute 'startswith'
Traceback (most recent call last):
  File "/home/esaevian/.local/lib/python3.7/site-packages/milc/milc.py", line 467, in __call__
    return self.__call__()
  File "/home/esaevian/.local/lib/python3.7/site-packages/milc/milc.py", line 472, in __call__
    return self._entrypoint(self)
  File "/home/esaevian/qmk_firmware/lib/python/qmk/cli/doctor.py", line 86, in doctor
    status = os_tests()
  File "/home/esaevian/qmk_firmware/lib/python/qmk/cli/doctor.py", line 26, in os_tests
    return os_test_linux()
  File "/home/esaevian/qmk_firmware/lib/python/qmk/cli/doctor.py", line 43, in os_test_linux
    if QMK_FIRMWARE.startswith("/mnt"):
AttributeError: 'PosixPath' object has no attribute 'startswith'
```

It seems [this line](https://github.com/qmk/qmk_firmware/commit/12f308748f3dd7ab2a1f3d93f8f207cb51e306a4#diff-d26098f964e0334af937ce9fee1a3dd91968877932f3370a6c7925964f9fe833R39) needs `QMK_FIRMWARE` cast to a string. I have this working on my local install.

Running Python 3.7.5

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: WSL / Ubuntu 18.04 / Win10
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version: preonic-1.0

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-04-02:
Whoops. It's supposed to have `.as_posix()` first. I had it in my WSL install I used to test it, but that's not where I committed it from. The above PR should fix it.

timovp, 2021-04-02:
Thanks for this. just ran into this when installing, and as the fix is small, this is was easily implemented untill PR is accepted :-)

PR #12466 (2021-04-02): `qmk doctor`: Fix WSL /mnt check

## Description

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #12465

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have made the appropr [... cut]
closed 2021-04-02 by zvecr