qmk/qmk_firmware #12476: [Bug] Toggling shift while holding a key assigned to an ANSI shifted keycode fails.

Wulfsta, 2021-04-03:
Holding an [ANSI shifted key](https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/keycodes_us_ansi_shifted) and toggling shift produces output like the following:
```
!!!!!!!!!!!!!!1111111111
              ^shift released here
```
@JonasGessner noticed this [here](https://github.com/qmk/qmk_firmware/pull/11422#issuecomment-809287212).

## System Information

 - Keyboard: HHKB Pro JP Type-S with Hasu Controller
 - Operating system: NixOS, Windows 10, Arch Linux (QMK built on this system), Ubuntu
 - AVR GCC version: 10.2.0

 - ARM GCC version: 10.2.0

 - QMK Firmware version: 0.11.69

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] Other: Arch Linux system is a laptop using the X windowing system's ability to remap keys to a Dvorak layout - I forget how I did this, and am testing across multiple machines/OSs anyway.

trudkin, 2021-04-16:
I am just an observer but from your description of the issue I would say working as intended/designed/specified.  This is similar to the caveats listed in the link provided. 

Key repeat is not a function of QMK but rather the OS. The example of KC_EXLM is not sent as **!** but rather **Shift-1**.   Or from the QMK point of view shift-down 1-down. When the shift is pressed it is still a shifted-1 but onced shift is released, shift-up, all that is left is 1.

What is the use case for this issue, as it might be a feature request.

JonasGessner, 2021-04-16:
> Key repeat is not a function of QMK but rather the OS. The example of KC_EXLM is not sent as ! but rather Shift-1. Or from the QMK point of view shift-down 1-down. When the shift is pressed it is still a shifted-1 but onced shift is released, shift-up, all that is left is 1.

I don't know, to me the behavior described by OP feels weird. Yeah I get why its happening but that doesn't make it right. It kind of goes against the idea of having macros like `KC_EXLM`. The idea of such macros should be that they behave just like regular keycodes and you shouldn't have to worry about what the individual keycodes are within that macro. The general principle should be "you hold the key and you get the output of that key, before and after holding mods". This principle holds for regular keycodes. For a macro like `KC_EXLM` this principle breaks, and you have to start to be aware of what the actual underlying keycodes are.

This specific issue would be easy to fix. The problem is that macros, like `KC_EXLM`, use `weak_mods` to register modifiers. Weak mods are unregistered on every key event which causes the behavior described by OP. Once shift is released the weak mods are cleared and the host OS only sees 1 being pressed. If macros explicitly used a different mod mask just for macros, which isn't cleared on every key event, this behavior would immediately disappear.

Unfortunately that wouldn't cover all cases. All you can do in a bitmask it set and unset bits, not keep a count. So pressing and holding two different macros that contain the same mod would still end up breaking in the same way.

Weak mods are being shared by a bunch of features already, each possibly messing with each others use of the same bitmask (same bitmask-is-not-a-counter thing). So it would make sense to move to a dedicated bitmask for macros from that perspective as well. Even if it could still produce weird behavior, at least it would do so in fewer cases and you'd have  one less feature using weak mods, which might alleviate other problems too.

trudkin, 2021-04-17:
The solution  is easy.  It is not the weak mods that are the issue.   It is the insertion of a direct SHIFT that is at issue.  So a solution  to eat the unregister of the shift.  The code can go in any process_record_XXX,  This should only be considered as proof of concept.

``` 
       static bool is_ansi;    
...
        switch (keycode) {
        case KC_LSFT: /*test other forms of shift also */
            if (record->event.pressed) {
                return true;
            } else {
                if (is_ansi) {
                    return false;
                } else {
                    return true;
                }
            }
            break;
        case KC_EXLM: /* include ansi range of codes */
            if (record->event.pressed) {
                is_ansi =true;
            } else {
                unregister_code(keycode & 0xff);
                unregister_code(KC_LSFT); /*should capture actual mod*/
                is_ansi =false;
            } 
            return is_ansi;
            break;
           ...
```
I still don't consider this a bug

IsaacElenbaas, 2021-04-17:
Weak mods are (should be?) only cleared on key down, no? If shift was pressed before this, and released after `KC_EXLM` was pressed, I agree that this is a bug. The weak mods should persist until an interrupt of a physical key, which would stop the keyrepeat and prevent any `1`s. If shift was tapped while `KC_EXLM` was keyrepeating, this isn't a QMK issue. It's the OS that decides to not stop keyrepeating upon a shift state change, which is bad design IMO.

IsaacElenbaas, 2021-04-17:
One could argue that weak mods should not be cleared upon a non-typing-key/modifier press, as there is no typing character happening that would be messed up by modifiers, but they are cleared before the keycode is determined, just before `action_tapping` IIRC. That isn't really feasable. We could probably store `stale_mods` or something and `add_weak_mods(stale_mods)` in `process_action`, but I doubt anyone wants yet another modifier bitmap and it would require a lot of testing to ensure Tap Holds aren't messed up.