qmk/qmk_firmware #12481: [Bug] LSFT + RSFT + <num> breaks layout

samhh, 2021-04-04:
Holding both shift keys and entering a number on the top row of the keyboard causes strange output / breaks layout. For example, LSFT + RSFT + 1 causes layer 1 to be permanently activated. Some other numbers do the same, some give other strange output, and some do nothing.

## System Information

 - Keyboard: Ergodox EZ
   - Revision: 5a 
 - Operating system: Arch Linux
 - AVR GCC version: 10.2.0

 - ARM GCC version: 10.2.0

 - QMK Firmware version: 0.12.32

 - Any keyboard related software installed? No.

## Additional Context

The relevant custom keymap if relevant: [samhh/qmk_firmware](https://github.com/samhh/qmk_firmware/tree/samhh/keyboards/ergodox_ez/keymaps/samhh)

elfmimi, 2021-04-06:
That behavior is by design. Here is its documentation: [COMMAND feature](https://docs.qmk.fm/#/feature_command)
Since you only have 3 layers defined, LSHIFT + RSHIFT + 3 and so forth give undefined results. 

You may disable those combinations by adding
```C
#define MAGIC_KEY_SWITCH_LAYER_WITH_FKEYS false
#define MAGIC_KEY_SWITCH_LAYER_WITH_NKEYS false
```
to your keymap's `config.h`

or otherwise you may add
```Makefile
COMMAND_ENABLE = no
```
to your keymap's `rules.mk` to disable the COMMAND feature completely.

samhh, 2021-04-06:
Clearly my Google-fu has failed me. :see_no_evil: 

Disabling the feature has solved my problem, many thanks!

closed 2021-04-06 by samhh