qmk/qmk_firmware #12501: Can't setup QMK in MacOS Big Sur

zk2004mb, 2021-04-06:
After installation in brew, the QMK can't be called.
Here is the error info:

(base) xxxxx@xxxxxx-iMac-Pro ~ % qmk
Error: %s: %s ('ImportError', ImportError("cannot import name 'format_ansi' from 'milc' (/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/milc/__init__.py)"))
Traceback (most recent call last):
  File "/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/qmk_cli/script_qmk.py", line 99, in main
    import qmk.cli  # noqa
  File "/Users/xxx/qmk_firmware/lib/python/qmk/cli/__init__.py", line 13, in <module>
    from . import doctor
  File "/Users/xxx/qmk_firmware/lib/python/qmk/cli/doctor.py", line 13, in <module>
    from qmk.questions import yesno
  File "/Users/xxx/qmk_firmware/lib/python/qmk/questions.py", line 4, in <module>
    from milc import cli, format_ansi
ImportError: cannot import name 'format_ansi' from 'milc' (/usr/local/Cellar/qmk/0.0.45/libexec/lib/python3.9/site-packages/milc/__init__.py)

System: MacOS Big Sur 11.2.1

zk2004mb, 2021-04-06:
Also when I check the milc package, it seems format_ansi has been removed.

Version 1.3.0
You can now set the program version number with os.environ['MILC_APP_VERSION'] = '1.2.3'
New global arguments: –unicode and –no-unicode
ANSIFormatter, ANSIStrippingFormatter, ANSIEmojiLoglevelFormatter, and ANSIStrippingEmojiLoglevelFormatter have been removed in favor of format_ansi() being color aware. The new MILCFormatter is being used in its place.

How should I deal with this?

jeroentbt, 2021-04-12:
Your `qmk_firmware` directory probably is out of date.
Try renaming it and run `qmk` again. If that works you're probably just behind on upstream.

zk2004mb, 2021-04-12:
> Your `qmk_firmware` directory probably is out of date.
> Try renaming it and run `qmk` again. If that works you're probably just behind on upstream.

Thank you! It worked.

closed 2021-04-12 by zk2004mb