qmk/qmk_firmware #12514: [Dumbpad v1x] Config.h for combo_teensy

mwsteb, 2021-04-08:
The `/qmk_firmware/keyboards/dumbpad/v1x/config.h` file defines pinout for Pro Micro chip. To make it work for Teensy2.0, different pins must be assigned.

This is what worked for me:

```
#define MATRIX_ROW_PINS { F5, F6, F7, B6 }
#define MATRIX_COL_PINS { D1, D2, D3, C6, C7 }
#define UNUSED_PINS

/* Single rotary encoder */
#define ENCODERS_PAD_A { D7 }
#define ENCODERS_PAD_B { D0 }

/* Onboard LEDs  */
// B4 LEFT
// B5 MIDDLE
// D6 RIGHT
#define LED_00 D6
#define LED_01 B5
#define LED_02 B4
```

Would it make sense to have different `config.h` files, for different chips? Maybe even split `v1x` directory into `v1x/combo` and `v1x/combo_teensy`?

Also, as it might be related to the pinout issue - I'm unable to actually make use of LEDs on my v1.1 combo_teensy board. I soldered in 3 x 3mm LEDs, with 82R resistor before each one. Using default keymap and `v1x.c`, at no point do those LEDs actually blink/shine.

Now, if I use my multimeter in continuity mode, attach red terminal to the D6 pin, and ground terminal to the cathode of the right LED (right LED, defined as LED_00), it shines. Same for other pins and LEDs. Which makes me think I haven't screwed up soldering.

Could there be something off with the D6 pin? It's connected to right LED on the PCB, but it also has it's own 1k resistor and a small LED on the teensy board. 

If I'm missing something obvious - I've just learned basics of soldering for this project, go easy on me :)

mwsteb, 2021-04-09:
@imchipwood is this a known issue?

imchipwood, 2021-04-09:
I've honestly never even tested the Teensy PCB layout, and can't really remember why I made it in the first place... :)

I'll try to take a look at the Teensy board file this weekend and get back to you on the LED problem.

As for splitting configs in qmk_firmware, I'm fine with that. I'd prefer to keep the existing folder structure so people don't wonder "where'd v1x go?!" the next time they compile or use the configurator..  so it would make more sense to create a `v1x_teensy` folder and leave the others alone.

mwsteb, 2021-04-09:
> I'll try to take a look at the Teensy board file this weekend and get back to you on the LED problem.

Thanks, I really appreciate it!

> As for splitting configs in qmk_firmware, I'm fine with that. I'd prefer to keep the existing folder structure so people don't wonder "where'd v1x go?!" the next time they compile or use the configurator.. so it would make more sense to create a v1x_teensy folder and leave the others alone.

I'll have a stab at it over the weekend.

closed 2021-08-21 by stale[bot]