qmk/qmk_firmware #12539: [Feature Request] Support for GMMK full-size keyboard

berzi, 2021-04-10:
## Feature Request Type

- [ X ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Support for GMMK TKL keyboards is there but not for 105-keys ones. If it doesn't prove too hard I'd be willing to work on support for it myself, but I don't know what needs to be done and where to start.

By the way, does a layout made for the TKL keyboard of the same brand work on a 105-key one (presumably without being able to remap numpad keys etc) or is it completely unusable?

WholesomeDucky, 2021-04-13:
While it is POSSIBLE that the firmware might flash, you should absolutely not try it. It's very likely to not work properly, since the PCB design may be different. I can't speak to how easy this would be implement based on the existing TKL support but I'll ask some questions today. If some of the more experienced folks think this would be easy to do, I could try to get my hands on a board and give it a shot.

berzi, 2021-04-14:
Thank you and sorry, it isn't actually the TKL but the GMMK Pro (which has a 75% form factor) that is compatible. I wrote to GMMK support and they say no other keyboard of theirs is compatible with QMK.
Hopefully it's still possible to make it work, but at this point I don't have any high hopes.

closed 2021-04-14 by berzi

CalcProgrammer1, 2021-09-30:
We have achieved basic QMK support for GMMK Full (which should also work for TKL) in our SonixQMK fork:

https://github.com/SonixQMK/qmk_firmware

Note that there are at least three known PCBs for the GMMK Full, which we're calling rev1, rev2, and rev3.  The rev1 is unsupported, apparently was an early model with single-color LEDs.  The rev2 uses the EVision VS12L03A LED driver chips and the rev3 uses the EVision VS12L17A LED driver chips.  The EVision VS11K13A microcontroller is the heart of all of the non-Pro GMMKs which is a rebranded Sonix SN32F268.  This is an ARM Cortex M0 microcontroller with 2KB of RAM and 32KB (30KB usable) of Flash.  Because of the very limited 2KB RAM for EEPROM emulation and the lack of proper EEPROM on the SN32 microcontrollers, we don't have VIA support.  The unofficial OpenRGB protocol does work, however.

I have made a tutorial video here:

https://www.youtube.com/watch?v=aUiKHdI5Vk8

The tutorial is a bit out of date now because we only had the newer PCB (rev3) working when I made it, but the flashing procedure is the same for both rev2 and rev3.  The same flashing tool and jumploader work for both, just the firmware needs changed.  If you flash the wrong firmware (rev2 on rev3 or vice versa) the keyboard will still work, just the LEDs won't function.  You can go back into bootloader and reflash the correct firmware.