qmk/qmk_firmware #12541: Add in support for OpenBSD [Feature Request]

WhiteGenghis, 2021-04-10:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
Requesting OpenBSD support for QMK.

0mp, 2021-04-29:
You may want to take a look at #12085 for inspiration.

fauxpark, 2021-04-29:
OpenBSD seems to have the required ARM and AVR toolchain packages, however they look like pretty old versions. I would look into updating them to at least GCC 8.

Erovia, 2021-05-04:
The more platform QMK is easily accessible on the happier we are.
Initial and continuous support needs community involvement, though.

If someone steps up and creates the necessary scripts and checks if all aspects are working we will be happy to merge it.
Given the tooling the project uses are widely used on Unix-like systems, I think supporting OpenBSD is just a matter of someone familiar with the platform doing the legwork.

closed 2021-09-03 by stale[bot]