qmk/qmk_firmware #12565: [Bug] kimiko: use BOOTLOADER = caterina with pro-micro board

trougnouf, 2021-04-12:
## Describe the Bug

`qmk flash -kb keycapsss/kimiko/rev1 -km default ` fails with `BOOTLOADER = atmel-dfu` (and dfu-programmer), requires `BOOTLOADER = caterina`

## System Information

 - Keyboard:  Keycapsss Kimiko rev1
 - Operating system: Arch Linux
 - AVR GCC version: 10.2
 - QMK Firmware version: 1a7a821582d3b848d50fddd6e942ed278bf36eff

## Additional Context

Flashing fails with the provided config. The keyboard is compatible with a pro micro board (which requires caterina or avrdude to work) and another board which has USB-C. Maybe the other board works with atmel-dfu?

The right bootloader should be set in [rules.mk](https://github.com/qmk/qmk_firmware/blob/master/keyboards/keycapsss/kimiko/rules.mk) and if user intervention is required then that should be documented.

@BenRoe

BenRoe, 2021-04-13:
Hello,
the flash command uses the BOOTLOADER setting in the rules.mk.
I have set the default to dfu, because most of the customer use the Puchi-C or a similar Pro Micro Clone with DFU Bootloader.  
If you use a Pro Micro with a  caterina bootloader, use `make keycapsss/kimiko:default:avrdude`, more infos in the [Kimiko Build-Guide](https://github.com/BenRoe/Kimiko-Build-Guide/blob/master/buildguide_en.md#flash-keymap-on-pro-micropuchi-c)

trougnouf, 2021-04-13:
Thank you @BenRoe . I didn't realize you had provided instructions in the build guide, the link that points down to it wasn't working due to an extra "/" (I sent a PR to fix it).

The Pro Micro I got in your shop works with caterina, I don't know if it changes per batch. I think it's probably better mentioning in the QMK README too but the information is already available so it's your call.

closed 2021-04-13 by trougnouf