qmk/qmk_firmware #12571: [Bug] Docker build not working when keyboard or keymap folder is a symlink

bomberstudios, 2021-04-13:
## Describe the Bug

What it says in the title: apparently, if a symlink is used for the keyboard or keymap folder, Docker won’t be able to reach the files and throw an error.

## Steps to reproduce

- add a symlink to a keyboard definition in the `keyboards` folder:
  [image]
- try to build it with Docker using `util/docker_build.sh z12`

### Expected result

- firmware should build

### Actual result

- qmk throws an error suggesting it can’t find the keyboard definition files:
  [image]

### Workaround

Copying the folder, instead of using a symlink, seems to work. But I would rather not do that, and keep my files symlinked.

Also, building the firmware locally without Docker works as expected.

## System Information
- Operating system: macOS 11.2.3 (20D91)
- Docker version:
  - cli: 20.10.5
  - Desktop: 3.3.0 (62916)

fauxpark, 2021-04-13:
Symlinks are not really supported in QMK, since there are issues with them on Windows. So the build system may or may not work with them.

bomberstudios, 2021-04-13:
Interesting, they seem to work reasonably well on macOS so that’s why I assumed they were supported.

Guess I need to find a better way to keep my files in order, thanks for your reply!

closed 2021-04-13 by bomberstudios

tzarc, 2021-04-13:
Just for reference, the docker container's view of the filesystem won't match the host, which is why the symlinks won't work.
You'd need to mount the volumes into the same absolute paths inside the container, overwriting the permissions on the host filesystem so as to allow access by the user inside the container.

Needlessly complex.