qmk/qmk_firmware #12593: [Feature Request] Add support for the STM32 STM32L433XX family of MCU

donicrosby, 2021-04-15:
## Feature Request Type

[ X ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)

## Description

I would like to add support for the STM32L433XX family that already has a `stm32_registry.h` in the ChibiOS, and has the necessary features need for basic QMK support. I was wondering what files I need to update in order to add support for this MCU.

When I tried to compile an initial version of my firmware to test some functionality with the MCU. The API told me it was not compatible. How would I go about adding support in both the API and the qmk binary if the MCU already has a valid `stm32_registry.h`? Is there a workflow/process I can follow then make a pull request?

tzarc, 2021-04-15:
See #12063.

tzarc, 2021-04-28:
Should be working, now.

closed 2021-08-28 by stale[bot]