qmk/qmk_firmware #12665: [Bug] KC_NUHS sending same keycode as KC_BSLS (51)

sss123next, 2021-04-23:
KC_NUHS / KC_NONUS_HASH sending same keycode as KC_BSLS, both sending code 51

```
KeyPress event, serial 32, synthetic NO, window 0x3e00001,
    root 0x723, subw 0x0, time 42720475, (305,673), root:(3185,695),
    state 0x0, keycode 51 (keysym 0x5c, backslash), same_screen YES,
    XLookupString gives 1 bytes: (5c) "\"
    XmbLookupString gives 1 bytes: (5c) "\"
    XFilterEvent returns: False

KeyRelease event, serial 35, synthetic NO, window 0x3e00001,
    root 0x723, subw 0x0, time 42720603, (305,673), root:(3185,695),
    state 0x0, keycode 51 (keysym 0x5c, backslash), same_screen YES,
    XLookupString gives 1 bytes: (5c) "\"
    XFilterEvent returns: False

KeyPress event, serial 35, synthetic NO, window 0x3e00001,
    root 0x723, subw 0x0, time 42721483, (305,673), root:(3185,695),
    state 0x0, keycode 51 (keysym 0x5c, backslash), same_screen YES,
    XLookupString gives 1 bytes: (5c) "\"
    XmbLookupString gives 1 bytes: (5c) "\"
    XFilterEvent returns: False

KeyRelease event, serial 35, synthetic NO, window 0x3e00001,
    root 0x723, subw 0x0, time 42721595, (305,673), root:(3185,695),
    state 0x0, keycode 51 (keysym 0x5c, backslash), same_screen YES,
    XLookupString gives 1 bytes: (5c) "\"
    XFilterEvent returns: False

KeyPress event, serial 35, synthetic NO, window 0x3e00001,
    root 0x723, subw 0x0, time 42725915, (305,673), root:(3185,695),
    state 0x0, keycode 133 (keysym 0xffeb, Super_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

## System Information

 - Keyboard:  melody96
 - Operating system: Linux  5.10.30
 - AVR GCC version: avr-gcc (GCC) 7.5.0

 - ARM GCC version: arm-none-eabi-gcc (GCC) 7.3.1 20180622 (7-2018-q2-update)

 - QMK Firmware version: 0.12.38

## Additional Context

drashna, 2021-05-14:
That's being reported by the OS, right? 

If so, then that's an issue with the host, because QMK is sending very different codes for backslash and non-US hash. 

You can see this in the enum for the keycodes.  Backslash is 0x31, and the non-US Hash is 0x32

https://github.com/qmk/qmk_firmware/blob/495d47b7d2a0df4e84c25c9cfee3931576cfbd32/tmk_core/common/keycode.h#L269-L271

sss123next, 2021-05-15:
yes, reported by os, you probably right, looks like os problem.

excuse for troubling.

sss123next, 2021-05-15:
hmm..., i probably should report it to xorg or linux... someday....

closed 2021-05-15 by sss123next