qmk/qmk_firmware #12666: [Bug] melody96 wrong keymap layout

sss123next, 2021-04-23:
incorrect keymap layout.

with current layout location of KC_ENT, KC_BSLS (near enter), KC_NUHS is shuffled

here is example of keymap which works properly with current qmk:

iso (layout D):

```
 [_default] = LAYOUT(
    KC_ESC,  KC_F1,   KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,  KC_F11,  KC_F12,  KC_INS, KC_HOME, KC_PGUP,  KC_PGDN, KC_END, KC_DEL,
    KC_GRV,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSPC, KC_BSPC, KC_NLCK, KC_PSLS, KC_PAST, KC_PMNS,
    KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_NONUS_HASH,          KC_P7,   KC_P8,   KC_P9,   KC_PMNS,
    KC_CAPS, KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT, KC_ENT,                    KC_P4,   KC_P5,   KC_P6,   KC_PPLS,
    KC_LSFT, KC_BSLS, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT,          KC_UP,   KC_P1,   KC_P2,   KC_P3,   KC_PENT,
    KC_LCTL, KC_LGUI, KC_LALT,                            KC_SPC,                             KC_RALT, MO(fn_base),   KC_LCTL,   KC_LEFT, KC_DOWN, KC_RGHT, KC_P0,   KC_PDOT, KC_PENT),
```

ansi (layout C):
```
     [_default] = LAYOUT(
    KC_ESC,  KC_F1,   KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,  KC_F11,  KC_F12,  KC_INS, KC_HOME, KC_PGUP,  KC_PGDN, KC_END, KC_DEL,
    KC_GRV,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSPC, KC_BSPC, KC_NLCK, KC_PSLS, KC_PAST, KC_PMNS,
    KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_ENT,          KC_P7,   KC_P8,   KC_P9,   KC_PMNS,
    KC_CAPS, KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT, KC_BSLS,                    KC_P4,   KC_P5,   KC_P6,   KC_PPLS,
    KC_LSFT, KC_NUBS, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT,          KC_UP,   KC_P1,   KC_P2,   KC_P3,   KC_PENT,
    KC_LCTL, KC_LGUI, KC_LALT,                            KC_SPC,                             KC_RALT, MO(fn_base),   KC_LCTL,   KC_LEFT, KC_DOWN, KC_RGHT, KC_P0,   KC_PDOT, KC_PENT),
```

## System Information

 - Keyboard: melody96
   - Revision (if applicable): hotswap layout D, hotswap layout C
 - Operating system: Linux 5.10.30
 - AVR GCC version: avr-gcc (GCC) 7.5.0

 - ARM GCC version: arm-none-eabi-gcc (GCC) 7.3.1 20180622 (7-2018-q2-update)

 - QMK Firmware version:  0.12.38

tpstevens, 2021-05-06:
Which version of the board do you have? Do you have a non-hotswap board that you're trying to use a hotswap layout on? There's also an older silver-pad PCB and a newer gold-pad PCB.

sss123next, 2021-05-07:
i have two pcb: hotswap layout D, hotswap layout C

sss123next, 2021-05-07:
used layouts names from https://www.aliexpress.com/item/32957159275.html

tpstevens, 2021-05-07:
Ah, that's because you should be using the LAYOUT_hotswap macro. The LAYOUT and LAYOUT_std60_split_num0 apply to the non-hotswap boards, and the LAYOUT_hotswap applies to the hotswap board. I'm working on a solution now: adding four macros to cover the four hotswap layouts and documenting the LAYOUT macro better to make it clear.

tpstevens, 2021-05-07:
YMDK had to reverse the ENTER and | keys for some reason, as documented in the Manuals section on the [melody96 page](https://ymdkey.com/products/kit-1-aluminum-cnc-top-acrylic-cnc-bottom-ymdk-melody-96-hotswap-pcb-keyboard-kit)

sss123next, 2021-05-07:
understood, it's not a problem for me at all, just want to let know about problem.

PR #22113 (2023-09-24): YMDK Melody96 Matrix Touch-Up
## Description

This PR fixes a long-standing bug regarding the ANSI Backslash and Enter keys. I meant to chase this down long ago.

- Correct layout/matrix diagram / Correct matrix data
  Swaps the matrix designations for the ANSI Backslash and ANSI Enter keys on Melody96 soldered PCBs, which has been broken since #16455.
- Implement `LAYOUT_hotswap` directly
  This is a temporary fix; I have a branch in progress to break up the support for Soldered and Hotswap PCBs into their own directories (the Hotswap PCB is multi-layout).
- Specify primary support for Solder [... cut]
closed 2023-09-24 by noroadsleft