qmk/qmk_firmware #12690: [Feature Request] some thinkpad media keycodes aren't supported

mvdan, 2021-04-25:
I have a Thinkpad X1 Carbon 6th gen. Here is a picture of its keyboard: 
[image]

Some media keys are supported directly, you can see them in https://docs.qmk.fm/#/keycodes. For example, F1/MUTE, F3/VOLUP, and F6/BRIGHTNESS_UP. Note that these keys are accessible via Fn; F5 is just the regular F5, but Fn+F5 is "brightness down".

However, many others are missing. I also used libinput on Linux to grab their names and keycodes:

* F4 KEY_F20 (190)
* F7 KEY_SWITCHVIDEOMODE (227)
* F8 KEY_WLAN (238)
* F9 KEY_CONFIG (171)
* F10 KEY_BLUETOOTH (237)
* F11 KEY_KEYBOARD (374)
* F12 KEY_BOOKMARKS (156)

I'm not sure why the name for F4 turns out that way, but it's clearly just "mute microphone" and it works well. It usually shows up as XF86AudioMicMute. Similarly, F7 usually shows up as XF86Display.

Could these keycodes be added? I'm using the Oryx web configurator for the ZSA Moonlander, and I'm trying to have the same media keys as the laptop keyboard. As far as I can tell, Oryx is a parallel fork of QMK, and they support the keycodes that you support, so I think this is the right place to bring this up :)

fauxpark, 2021-04-25:
Linux "keycodes" are different to HID usages (what USB keyboards actually send), and are useless here unless you know what they map to, if any. It's likely that some functionality (in particular, the Fn key) is implemented in ACPI instead, so there might not really be any way to send them using QMK or any other HID device.

[`KEY_F20` is `KC_F20`](https://elixir.bootlin.com/linux/v5.11.16/source/drivers/hid/hid-input.c#L34).
[`KEY_BOOKMARKS`](https://elixir.bootlin.com/linux/v5.11.16/source/drivers/hid/hid-input.c#L972) is [`KC_WFAV`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/report.h#L266-L267).
[`KEY_CONFIG`](https://elixir.bootlin.com/linux/v5.11.16/source/drivers/hid/hid-input.c#L973) is [`KC_MSEL`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/report.h#L242-L243).

[`KEY_BLUETOOTH` and `KEY_WLAN` are not mapped to anything in `hid-input.c`](https://elixir.bootlin.com/linux/v5.11.16/C/ident/KEY_BLUETOOTH); you can see they are referenced in `thinkpad_acpi.c` though.

[`KEY_KEYBOARD`](https://elixir.bootlin.com/linux/v5.11.16/source/drivers/hid/hid-input.c#L1002) can be sent in a custom keycode by calling [`host_consumer_send(AL_KEYBOARD_LAYOUT)`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/report.h#L81) on keydown, and `0` on keyup.
[`KEY_SWITCHVIDEOMODE`](https://elixir.bootlin.com/linux/v5.11.16/source/drivers/hid/hid-input.c#L681) can likely be sent the same way, but with `host_system_send(0xB5)` instead (feel free to add a define for this in [`report.h`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/report.h#L97-L102)).

mvdan, 2021-04-25:
Thanks for your quick response! I wasn't aware that Linux keycodes were a different concept than firmware keycodes, so apologies for the confused feature request.

I was using SWITCHVIDEOMODE as a shortcut for a command, so I guess using CONFIG or BOOKMARKS is easier as I can trivially send that from the external keyboard too.

I don't think there's much to do here in terms of "support thinkpad media keys out of the box", so I'll close this.

closed 2021-04-25 by mvdan