qmk/qmk_firmware #12732: [Bug] Keyboard does not send keystrokes till it is plugged back in

obrian33, 2021-04-29:
Once computer goes to sleep and is awake the keyboard doesn't send keystrokes.

## Describe the Bug
After upgrading to MacOS Big Sur 11.3, when the M1 Mac Mini goes to sleep the keyboard doesn't send key strokes until replugging in the board. 

## System Information

 - Keyboard:  
 - Planck EZ Glow 
   - Revision (if applicable):  
 - Operating system:
 - MacOS Big Sur 11.3
 - AVR GCC version: 
 - avr-gcc (Homebrew AVR GCC 8.4.0_1) 8.4.0

 - ARM GCC version: 
 - arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version:
 - 0.12.32

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

obedparla, 2021-05-05:
I have the exact same issue.

- Ergodox EZ (no glow)
- MacOS Big Sur 11.2.3 (20D91)
- Not sure how to check the QMK firmware version tho.

kylpo, 2021-06-18:
Happens to me on 3 separate QMK keyboards, on both of my new M1 Macbook Airs.

Standard keyboards from Apple and Logitech seem to not have this issue.

mike1808, 2021-06-23:
I defined  `NO_USB_STARTUP_CHECK` in my config and it helped to solve the issue with GMMK Pro.

obedparla, 2021-06-23:
Thank you @mike1808 ! I changed it in Oryx for the Ergodox EZ and it fixed the issue: 
[image]

Erbenos, 2024-05-09:
I am still experiencing this exact issue on 

Keyboard:                Keychron Q4
System Version:	macOS 14.4.1 (23E224)

With v1 firmware. Keyboard does wake the PC from sleep, but upon waking the PC from sleep it doesn't send any keypresses to it. Setting NO_USB_STARTUP_CHECK is an unfortunate workaround, cause it disables the ability to wake PC from sleep altogether, though it does solve the initial issue.

Erbenos, 2024-05-09:
Setting the USB_SUSPEND_WAKEUP_DELAY as documented at https://docs.qmk.fm/#/config_options?id=hardware-options seems to work as a better workaround - keyboard does wake up the PC and it sends the keypresses after waking up too.