qmk/qmk_firmware #12758: Adding audio to the Drop CTRL [Bug]

mikegwatson, 2021-05-01:
## Describe the Bug
I've wired up a low profile piezo/speaker(p/n: AST1109MLQ) to pin PB13 of the Drop CTRL SAMD51J18A uC, but am running into QMK compile errors when using `AUDIO_ENABLE = yes` in rules.mk.  I've included `#define AUDIO_PIN B13` in config.h, but perhaps I'm missing something critical...

I receive the following error message:
> Compiling: keyboards/massdrop/ctrl/matrix.c                                                        In file included from quantum/audio/voices.h:25,
                 from quantum/audio/audio.h:23,
                 from quantum/quantum.h:85,
                 from keyboards/massdrop/ctrl/ctrl.h:3,
                 from keyboards/massdrop/ctrl/matrix.c:18:
quantum/audio/luts.h:24:14: fatal error: ch.h: No such file or directory
\#    include <ch.h>
              ^~~~~~

At face value the compiler doesn't seem to be able to find the file ch.h.  Any thoughts or feedback as to why it's not able to find this file.  I'm guessing I haven't configured something correctly or am missing a configuration.

Cheers,
Mike

## System Information

 - Keyboard: MassDrop CTRL 
 - Operating system: MacOS
 - AVR GCC version: 8.4.0
 - ARM GCC version: 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.12.29
 - Any keyboard related software installed? 
   - [X] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

tzarc, 2021-05-01:
Audio is impossible on the Massdrop boards without a significant amount of rework, as they produced their own hardware abstraction layer instead of using the current ARM abstraction layer -- ChibiOS.

ChibiOS is what provides the `ch.h` file -- as such, it's unable to be used with their boards.

QMK is unlikely to entertain any attempt at modifying their abstractions, due to the poor state they're in -- see #11119.

mikegwatson, 2021-05-01:
Oh wow!  Thank you very much for the explanation @tzarc, that would have taken me quite a while to unravel on my own!

closed 2022-08-28 by zvecr