qmk/qmk_firmware #12773: [Feature Request] Now that userspaces are a thing, add *_keymap versions of functions (e.g., process_record)

rmwphd, 2021-05-01:
In the `process_record_*` function group, there exist process_record_quantum, process_record_kb, and process_record_user, and (if the user wants to do some fun weak attribution stuff) process_record_keymap. Same with `encoder_update` and a bunch of others. In my experience, some of them actually seem to work _without_ screwing around with the workaround you can find [here](https://beta.docs.qmk.fm/using-qmk/software-features/feature_userspace), others don't. 

Putting _user functions in the keymap (rather than exclusively in the userspace) seems to clash with the paradigm established by the _quantum, _kb, _user hierarchy of allowing these various definitions to coexist naturally and play nice. Thus the feature request:

Please let's add _keymap versions of all of the functions that have _user versions and encourage their use in keymap files from now on, encouraging users to put _user versions of those functions in their userspaces.

drashna, 2021-05-15:
While userspace does exist (and has for a long while now), I don't agree.   Most people don't use the feature, and this adds additional complexity that non-userspace users may get tripped up on (eg, they may break it, if they move to userspace configs).  

However, the documentation for the Userspace feature does include these functions, as an example. 
https://docs.qmk.fm/#/feature_userspace?id=customized-functions

rmwphd, 2021-05-17:
Right, I linked that page's workaround in my initial FR. Mostly, this feature would be intended to improve consistency (only `_user` functions in user space, only `_keymap` functions in keymaps). If consistency isn't a roadmap priority (and since there's clearly a workaround available for those who want to use these features), then I'd agree with you that this request shouldn't eat up resources needed by other features in development.