qmk/qmk_firmware #12787: [BUG] Cannot build Whitefox Firmware

trashcatt, 2021-05-03:
## Describe the Bug

I cannot build any Whitefox keymaps I always get error

`Compiling: quantum/command.c                                                                        [OK]
Compiling: tmk_core/common/chibios/timer.c                                                         tmk_core/common/chibios/timer.c: In function 'timer_read32':
tmk_core/common/chibios/timer.c:41:22: error: implicit declaration of function 'TIME_I2MS' [-Werror=implicit-function-declaration]
   41 |     return (uint32_t)TIME_I2MS(systime - reset_point);
      |                      ^~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:412: .build/obj_whitefox_trashcat/common/chibios/timer.o] Error 1
Make finished with errors
make: *** [Makefile:523: whitefox:trashcat] Error 1`

## System Information

 - Keyboard:  Whitefox
   - Revision (if applicable):  
 - Operating system: Artix Linux
 - AVR GCC version: 8.3.0

 - ARM GCC version: 10.3.0

 - QMK Firmware version: 0.11.68

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2021-05-15:
Make sure that your repo is up to date, and try recompiling.

trashcatt, 2021-05-18:
Yup, it was something with my fork. Fixed. Thanks.

closed 2021-05-18 by trashcatt