qmk/qmk_firmware #12794: Pertition to remove brew upgrade from install script

brorbw, 2021-05-03:
The line to upgrade brew in the install script for macOS can have consequences for users. In my case it broke some of my installs. I can see that it ignores pinned versions but to be honest, nobody really uses this feature.

If this is not doable - there should be a promt to inform the user that the script will run brew upgrade and the ability to opt out!

The line in question:
[https://github.com/qmk/qmk_firmware/blob/master/util/install/macos.sh#L12](https://github.com/qmk/qmk_firmware/blob/master/util/install/macos.sh#L12)

askreet, 2021-05-04:
I just ran into this as well, big +1 from me. My local dev environment for work has some packages pinned and this nearly caught me by surprise. Fortunately when I saw it was updating _45 packages from homebrew_ I aborted it and started digging.

askreet, 2021-05-04:
In case anyone else lands here and is curious, this seems to be enough to get buildable on `master`:

```
brew update
brew upgrade qmk/qmk/qmk # or presumably install if fresh
brew link --force avr-gcc@8
brew link --force arm-gcc-bin@8
python3 -m pip install -r $QMK_FIRMWARE_DIR/requirements.txt
```

bb2020, 2021-05-16:
+1
It can be replaced with `brew update && brew outdated` just to remind about outdated packages, but without upgrading them.

fauxpark, 2021-05-16:
If running `brew upgrade` breaks your install, then you might have bigger problems.

The problem is that some of our dependencies *must* be relatively up to date before attempting install, so I don't see this line going anywhere anytime soon.

brorbw, 2021-05-16:
@fauxpark not really - `brew` is not that smart. I would suggest updating the specific dependencies and not everything if it's really necessary. Or checking the version of the dependencies and advising the user to update.

skullydazed, 2021-05-16:
We're facing several problems here and there's no good way to make everyone happy. Let's talk about the issues we're facing.

To start with howebrew is really not designed to "stay back". Yes, you can pin versions but there are situations where those versions will still be upgraded (like when they are a dependency for another program that is being upgraded.) Another problem is that you can very easily find yourself in a state that can't be rebuilt if you have a problem that requires moving to a new machine. These are not QMK specific but they are issues you should consider before pinning homebrew packages.

For an issue more relevant to QMK we have the support problem. We've found that every time we make a step optional we get more people popping up on discord asking why something is broken after they said no. As an entirely volunteer-driven project we are very sensitive to anything that takes volunteer time away from things like reviewing PRs. 

Then there's my personal experience- over my career I've spent too many hours dealing with conflicting packages and broken servers after some software package updated everything for me without notice. Believe me when I say I've experienced your pain and am re-experiencing it now second hand. It would be my preference that we never require the user to update before installing the software necessary for QMK.

This brings us to the current situation- why are we running `brew upgrade` even though many of the volunteers within QMK would prefer not to? Because it causes the fewest problems all around.

For everyone in your position there are easily 5 or 10 people who have avoided problems because that `brew upgrade` is there. In this specific case we are [following one of homebrew's philosophies](https://www.quora.com/Whats-the-logic-behind-Google-rejecting-Max-Howell-the-author-of-Homebrew-for-not-being-able-to-invert-a-binary-tree):

> it forces a lot of opinionated decisions on users if the maintainers think it benefits the majority

While we are not likely to change the core decision (if you feel you have a particularly strong argument we are always willing to hear it out) there is always room for improvement. The messaging in the tutorial and/or script can be improved as we work to find the right balance between presenting all relevant information and avoiding a wall of text that is completely ignored. There are possibly other changes that can improve the experience for people who are impacted by decisions that affect a minority of users.

It's clear this is an itch for some of you, I encourage you to start to scratch it with PR's that improve the situation without disrupting how it works for most users.

brorbw, 2021-05-16:
I've started a PR. I intended to open a draft PR but it appears that I can't do that. Anyhow I want to welcome anyone to comment on the PR.

@skullydazed thank you for your input. In regards to the "brew philosophies" there is a good saying were I'm from that _you should not wash your hands in someone else's filth_.

brorbw, 2021-08-22:
https://github.com/qmk/qmk_firmware/pull/12921

PR #12921 (2021-08-22): Only upgrade formulae ~~Remove brew upgrade and prompts the user~~
This will remove the old way of forcefully upgrading peoples systems
without proper notice and consent.

If there are any packages or dependencies that needs upgrading it will
prompt the user and he/she/they/ze can consent to the upgrade or
continue without at own risk.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #12794
## Che [... cut]
closed 2021-08-22 by skullydazed