qmk/qmk_firmware #12843: Cherry-pick Teensy 4.1 support into qmk/ChibiOS-Contrib?

stapelberg, 2021-05-09:
Could we please cherry-pick the following 3 merge commits into https://github.com/qmk/ChibiOS-Contrib?

1. https://github.com/ChibiOS/ChibiOS-Contrib/commit/b3774246770bcbc40ff0ab193ce2f72d707cf36d
1. https://github.com/ChibiOS/ChibiOS-Contrib/commit/cec80982e4ece77d14271fe374b1b6c459b5b33f
1. https://github.com/ChibiOS/ChibiOS-Contrib/commit/b387cc84d75ac1846f443c1a48f22f5d5de5dabe

These add support for the Teensy 4.1 micro controller, so they’re a strict addition, without the risk of breaking any existing controllers.

This support is a requirement for [my `kint41` controller](https://github.com/kinx-project/kint).

For the last controller support I have added (`kint36` with the Teensy 3.6), it took quite a while to wait for the changes to ChibiOS-Contrib to trickle down to the qmk repository, so I figured I’d ask this time to skip the lengthy waiting period :)

Thanks in advance

cc @tzarc

tzarc, 2021-05-09:
Will get updated next time we update the downstream repos. Should be this coming develop merge cycle.

stapelberg, 2021-05-09:
> Will get updated next time we update the downstream repos. Should be this coming develop merge cycle.

Ah, thanks! On https://beta.docs.qmk.fm/developing-qmk/breaking-changes/breaking_changes, I see May 29th listed as the end of that cycle with “Merge develop to master”.

Looks like May 29th is when we can proceed with PRs to develop, then :)

stapelberg, 2021-05-30:
So May 29th was yesterday and I do see a bunch of “2021 May 29 Breaking Changes” commits, but ChibiOS and ChibiOS-Contrib still seem to be at their old versions.

Any word on updating ChibiOS-Contrib to latest, or alternatively cherry-picking the commits we need the kint41 controller?

Thanks

tzarc, 2021-05-31:
Already doing a bunch of sanity checks. Base ChibiOS 20.3.3 upgrade as well as latest ChibiOS-Contrib coming.

tzarc, 2021-05-31:
Using: https://github.com/tzarc/ChibiOS-Contrib/tree/9c2bfa6aeba993345f5425d82807c101f8e25e65

tzarc, 2021-05-31:
Re-reading, I think there was a misinterpretation of my comment -- ChibiOS and ChibiOS-Contrib upgrades happen during a `develop` merge cycle, and for stability's sake we chose not to upgrade them during the last cycle.

`develop` will be picking up the new ChibiOS and ChibiOS-Contrib, with the intent of using the next three months to test compatibility.

stapelberg, 2021-05-31:
Thanks for the status update! Getting the ChibiOS-Contrib tree you linked into the `develop` branch works for me :)

tzarc, 2021-05-31:
Good to hear! Will be attempting to get it into normal QMK submodules over the next couple of days, and the corresponding `develop` PR so that it's locked in will follow.

tzarc, 2021-05-31:
`develop` should be good to go!

stapelberg, 2021-06-02:
Can confirm this works, I have rebased my changes on top of your update commits at https://github.com/kinx-project/qmk_firmware/commits/kint41-upstream and will work to send pull requests now. Thank you!

PR #13056 (2021-05-31): Update ChibiOS, ChibiOS-Contrib.
## Description

Updates to ChibiOS 20.3.3, and the latest ChibiOS-Contrib matching the 20.3.x branch.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Closes #12843

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [* [... cut]
closed 2021-06-02 by stapelberg