qmk/qmk_firmware #12862: [Bug] kinesis/kint does not compile default keymap

jfrankl, 2021-05-11:
## Describe the Bug

The `kinesis/kint36` will not compile with the default keymap. I am starting with a fresh install of QMK on the `develop` branch. When I run `qmk compile -kb kinesis/kint36 -km default`, I receive the following error: 

```
Linking: .build/kinesis_kint36_default.elf                                                          [WARNINGS]
 |
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:285: warning: memory region `flash4' not declared
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:293: warning: memory region `flash5' not declared
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:301: warning: memory region `flash6' not declared
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:310: warning: memory region `flash7' not declared
 |
Creating binary load file for flashing: .build/kinesis_kint36_default.bin                           [OK]
sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [.build/kinesis_kint36_default.bin] Error 2
make: *** [kinesis/kint36:default] Error 1
Make finished with errors
```

## System Information

 - Keyboard:  kinesis/kint36 
 - Operating system: Mac OS 11.2.3
 - AVR GCC version:  8.4.0

 - ARM GCC version: 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.12.48

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

@stapelberg I thought you might be interested in this.

fauxpark, 2021-05-11:
For whatever reason, the `MK66FX1M0` ldscript only specifies `flash0` through `flash3`:
https://github.com/chibios/ChibiOS-Contrib/blob/chibios-20.3.x/os/common/startup/ARMCMx/compilers/GCC/ld/MK66FX1M0.ld#L32

Adding the following lines to it seems to get the linker to succeed:
```
  flash4   : org = 0x00000000, len = 0
  flash5   : org = 0x00000000, len = 0
  flash6   : org = 0x00000000, len = 0
  flash7   : org = 0x00000000, len = 0
```

There appears to be something wrong with the make logic on `develop` and I actually get a syntax error afterwards, but I believe this to be unrelated as it also happens when compiling for STM32.

stapelberg, 2021-05-11:
I thought the linker only threw warnings regarding the flash regions, not errors? 

Either way, it would be good to fix that, warning or not :)

Perhaps @jfrankl can try it out and report back. Also, could you clarify whether the image is actually built (and the failure merely cosmetic)?

fauxpark, 2021-05-11:
Yeah, they are warnings, the linker is not actually failing here - I just noticed the unrelated syntax error I was talking about is in OP.

stapelberg, 2021-05-11:
> For ARM MCUs, the output filetype is `.bin`. Only AVR firmware is converted to `.hex`.

No, the Teensy MCUs (newer models are ARM-based) also use .hex files with their `teensy-loader-cli` program.

fauxpark, 2021-05-11:
Sure, but in the context of QMK, the rule of thumb is ARM = bin, AVR = hex.

jfrankl, 2021-05-11:
Thanks everyone for taking a look at this. I'm amazed by how quickly this got solved.

PR #12866 (2021-05-11): Fix syntax error when compiling for ARM

## Description

Seems to be macOS specific - something to do with zsh?
```
Creating binary load file for flashing: .build/planck_rev6_default.bin                              [OK]
sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [.build/planck_rev6_default.bin] Error 2
make: *** [planck/rev6:default] Error 1
Make finished with errors
```

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation
 [... cut]