qmk/qmk_firmware #12863: Issues with lighting layers for a Planck Rev 6

HunterArmstrong, 2021-05-11:
Currently I am running into issues with the RGB lighting layer configuration in MSYS for my planck rev6.

I followed the instructions in the QMK doc online and typed out this onto my keymap.c file:
#define RGBLIGHT_LAYERS

const rgblight_segment_t PROGMEM my_Function_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {0, 9, HSV_WHITE}
);
const rgblight_segment_t PROGMEM my_Numpad_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {0, 9, HSV_BLUE}
);
const rgblight_segment_t PROGMEM my_Gaming_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {0, 9, HSV_GREEN}
);
const rgblight_segment_t PROGMEM my_Sym_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {0, 9, HSV_RED}
);

const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(
    my_Function_layer,
    my_Numpad_layer,    
    my_Gaming_layer,    
    my_Sym_layer     
);

void keyboard_post_init_user(void) {
    // Enable the LED layers
    rgblight_layers = my_rgb_layers;
}
One compile error later, and i got the same three errors for each layer in my keymap file.
keyboards/planck/keymaps/2SpacePlanck/keymap.c:95:7: error: unknown type name 'rgblight_segment_t'

   95 | const rgblight_segment_t PROGMEM my_Function_layer[] = RGBLIGHT_LAYER_SEGMENTS(
      |       ^~~~~~~~~~~~~~~~~~

keyboards/planck/keymaps/2SpacePlanck/keymap.c:95:56: error: implicit declaration of function 'RGBLIGHT_LAYER_SEGMENTS'; did you mean 'RGBLIGHT_LAYERS'? [-Werror=implicit-function-declaration]
   95 | const rgblight_segment_t PROGMEM my_Function_layer[] = RGBLIGHT_LAYER_SEGMENTS(
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                        RGBLIGHT_LAYERS

keyboards/planck/keymaps/2SpacePlanck/keymap.c:96:5: error: expected expression before '{' token
   96 |     {0, 9, HSV_WHITE}
      |     ^

Did I type something wrong into the keymap file or was there a step that I missed altogether?

I really need help here

drashna, 2021-05-15:
It sounds like you're missing the define for the `RGBLIGHT_LAYERS`. Make sure that `#define RGBLIGHT_LAYERS` exists in your keymap's config.h file.

HunterArmstrong, 2021-05-15:
Fixed it. Thank you 😁😁

closed 2021-09-21 by stale[bot]