qmk/qmk_firmware #12880: [Bug] Function `get_mods` returning mirrored key on split keyboards

hexcowboy, 2021-05-12:
The `get_mods` function is returning incorrect values for keypresses on split keyboards (tested, crkbd).

## Describe the Bug

The function is returning modifier keys pressed for a mirrored key. For example, if my LEFT CTRL is mapped to the thumb key on the left shield, `get_mods` will return nothing when that key is pressed, but it will return LEFT CTRL when the left thumb key is pressed on the right shield.

## System Information

 - Keyboard:  crkbd
   - Revision:  rev 3
 - Board: Elite C
 - Operating system: macOS
 - AVR GCC version:  avr-gcc (Homebrew AVR GCC 8.4.0_1) 8.4.0
 - ARM GCC version:  arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.12.48

## Additional Context

This can be reproduced with `crkbd/gotham` which contains the [following code](https://github.com/qmk/qmk_firmware/blob/9a3f4f8f8a3090b3783dfd60a75cd8120ec30e65/keyboards/crkbd/keymaps/gotham/oled.c#L119) which displays modifier key pressed on the OLED:
```c
void render_mod_status(void) {
#ifdef NO_ACTION_ONESHOT
    uint8_t modifiers = get_mods();
#else
    uint8_t modifiers = get_mods() | get_oneshot_mods();
#endif

    (modifiers & MOD_MASK_CTRL) ? render_mod_ctrl() : oled_write_P(PSTR("  "), false);
    oled_write_P(PSTR(" "), false);
    (modifiers & MOD_MASK_SHIFT) ? render_mod_shift() : oled_write_P(PSTR("  "), false);

    (modifiers & MOD_MASK_ALT) ? render_mod_alt() : oled_write_P(PSTR("  "), false);
    oled_write_P(PSTR(" "), false);
    (modifiers & MOD_MASK_GUI) ? render_mod_gui() : oled_write_P(PSTR("  "), false);
}
```
And this keymap:
```c
  [_QWERTY] = LAYOUT(
  //,-----------------------------------------------------.                    ,-----------------------------------------------------.
       KC_ESC,    KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,                         KC_Y,    KC_U,    KC_I,    KC_O, KC_SCLN, KC_BSPC,
  //|--------+--------+--------+--------+--------+--------|                    |--------+--------+--------+--------+--------+--------|
       KC_TAB,    KC_A,    KC_S,    KC_D,    KC_F,    KC_G,                         KC_H,    KC_J,    KC_K,    KC_L,    KC_P,  KC_ENT,
  //|--------+--------+--------+--------+--------+--------|                    |--------+--------+--------+--------+--------+--------|
      KC_LALT,    KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,                         KC_N,    KC_M, KC_COMM,  KC_DOT, KC_SLSH, KC_RALT,
  //|--------+--------+--------+--------+--------+--------+--------|  |--------+--------+--------+--------+--------+--------+--------|
                                          KC_LCTL, KC_LGUI, KC_LSFT,     KC_SPC,   MO(1),   MO(2)
                                      //`--------------------------'  `--------------------------'
  ),

```
Built with `make crkbd:gotham:dfu`

@GauthamYerroju

GauthamYerroju, 2021-05-12:
I'll need a couple of days to make time for this.

hexcowboy, 2021-05-12:
I should also note that a similar issue exists with the RGB matrix, the lights only change colors on opposite shields when the HI/LO layer is selected (with `RGB_MATRIX_ENABLE = yes` in the rules).

hexcowboy, 2021-05-13:
This issue is caused by `#define EE_HANDS` in config.h. Disabling it by removing this line should solve any problems.

closed 2021-05-13 by hexcowboy