qmk/qmk_firmware #12896: Requesting permission to use logo-background

david-janssen, 2021-05-14:
Hello,

I am the leading developer working on [kmonad](https://github.com/kmonad/kmonad), a project that tries to support a lot of QMK functionality in software, so that you can have weird keymaps on laptops and such.  We recently started brainstorming a logo. It'll probably be something like a stylized Greek Kappa, but some of my contributors suggested using the chip-background from the QMK logo as an homage to what first inspired our project. I am popping in to check if that was OK with you guys. I wouldn't want to step on anyone's toes. If you'd rather we didn't, for any reason whatsoever, we won't use it.

Thanks for the time,

David

jackhumbert, 2021-05-14:
Awesome! I think we'd appreciate it if you could differentiate the chip's design from ours (in shape, colors, etc), since we use our specific background in a lot of official QMK projects, but otherwise it's totally fine!

david-janssen, 2021-05-15:
Noted, and thanks!

closed 2021-05-15 by david-janssen